PX4/PX4-Autopilot #20843: Microdds_client disconnects with mRo Control Zero H7

gorghino, 2022-12-30:
Hi,
I flashed the latest PX4 repo in my mRo Control Zero H7. (synced with this PR (https://github.com/PX4/PX4-Autopilot/pull/20812)

In my SD card I put ```microdds_client start -t serial -d /dev/ttyS1 -b 115200``` in extras.txt.

On my Nvidia NX (Jetpack 5.0.2 - Ubuntu 20.04) I installed ROS2 humble by sources and I cloned micro-ROS-Agent in my workspace. No errors with compiling everything.

When I run ```./build/micro_ros_agent/micro_ros_agent serial -D /dev/ttyTHS1 -b 115200``` I can see it starting and subscribing the topics from PX4 but after 2/3sec the client is deleted and the session is closed. Then the session is either established or re-established.

```nvidia@nvidia-desktop:~/drone-racing$ ./build/micro_ros_agent/micro_ros_agent serial -D /dev/ttyTHS1 -b 115200
[1672233093.421832] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1672233093.422874] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1672233104.307837] info     | Root.cpp           | create_client            | create                 | client_key: 0x00000001, session_id: 0x81
[1672233104.308323] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x00000001, address: 1
[1672233104.395390] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x00000001, participant_id: 0x001(1)
[1672233104.414770] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E8(2), participant_id: 0x001(1)
[1672233104.415100] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E8(4), participant_id: 0x001(1)
[1672233104.416100] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E8(6), subscriber_id: 0x3E8(4)
[1672233104.439207] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E9(2), participant_id: 0x001(1)
[1672233104.439447] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E9(4), participant_id: 0x001(1)
[1672233104.440339] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E9(6), subscriber_id: 0x3E9(4)
[1672233104.463996] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EA(2), participant_id: 0x001(1)
[1672233104.464295] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EA(4), participant_id: 0x001(1)
[1672233104.465774] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EA(6), subscriber_id: 0x3EA(4)
[1672233104.487957] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EB(2), participant_id: 0x001(1)
[1672233104.488238] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EB(4), participant_id: 0x001(1)
[1672233104.489680] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EB(6), subscriber_id: 0x3EB(4)
[1672233104.517728] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0A1(2), participant_id: 0x001(1)
[1672233104.518132] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0A1(3), participant_id: 0x001(1)
[1672233104.519725] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0A1(5), publisher_id: 0x0A1(3)
[1672233104.544851] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0A8(2), participant_id: 0x001(1)
[1672233104.545183] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0A8(3), participant_id: 0x001(1)
[1672233104.546852] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0A8(5), publisher_id: 0x0A8(3)
[1672233104.572646] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x009(2), participant_id: 0x001(1)
[1672233104.573022] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x009(3), participant_id: 0x001(1)
[1672233104.574964] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x009(5), publisher_id: 0x009(3)
[1672233104.866546] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0AC(2), participant_id: 0x001(1)
[1672233104.867773] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0AC(3), participant_id: 0x001(1)
[1672233104.873510] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0AC(5), publisher_id: 0x0AC(3)
[1672233106.198530] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x0DE(2), participant_id: 0x001(1)
[1672233106.199584] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x00000001, publisher_id: 0x0DE(3), participant_id: 0x001(1)
[1672233106.204974] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x00000001, datawriter_id: 0x0DE(5), publisher_id: 0x0DE(3)
[1672233106.438109] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x00000001
[1672233106.438547] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x00000001, address: 1
[1672233106.439388] info     | Root.cpp           | create_client            | create                 | client_key: 0x00000001, session_id: 0x81
[1672233106.439504] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x00000001, address: 1
[1672233106.491187] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x00000001, participant_id: 0x001(1)
[1672233106.510310] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E8(2), participant_id: 0x001(1)
[1672233106.510564] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E8(4), participant_id: 0x001(1)
[1672233106.511468] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E8(6), subscriber_id: 0x3E8(4)
[1672233106.536030] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3E9(2), participant_id: 0x001(1)
[1672233106.537162] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3E9(4), participant_id: 0x001(1)
[1672233106.539827] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3E9(6), subscriber_id: 0x3E9(4)
[1672233106.561140] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x00000001, topic_id: 0x3EA(2), participant_id: 0x001(1)
[1672233106.561386] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x00000001, subscriber_id: 0x3EA(4), participant_id: 0x001(1)
[1672233106.562351] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x00000001, datareader_id: 0x3EA(6), subscriber_id: 0x3EA(4)
[... cut ...]
## Log Files and Screenshots
"Flight" log file: https://logs.px4.io/plot_app?log=8e5d71c7-3d26-4630-9a0a-30b70c9780d1
micro-ros-agent LOG with v6 flag: [micro-ros-agent-v6-mro.log](https://github.com/micro-ROS/micro-ROS-Agent/files/10314511/micro-ros-agent-v6-mro.log)

[image]

**System information (please complete the following information):**
 - OS: Ubuntu 20.04 - NVIDIA NX (Carrier board) - Jetpack 5.0.2
 - ROS 2 Humble (Tested also with Foxy)
 - Micro-ros-Agent Version: 0e41ea652238bd0123f6249d69b533d282882dd6

gorghino, 2023-01-02:
Increasing the packet loss counter in the ping check, fixed the problem.

I set it to 5.
https://github.com/PX4/PX4-Autopilot/blob/f2c71a887448935c6f3294dd60b1d17d37237374/src/modules/microdds_client/microdds_client.cpp#L363

It's more a workaround than a proper fix. In case you need further investigation, please let me know

russkel, 2023-01-11:
I am also experiencing this issue.

dagar, 2023-01-12:
I'll get setup to debug this, so far I've mainly used microdds_client in SITL and on real hardware over USB.

russkel, 2023-01-12:
I have too used it from SITL and these issues _usually_ did not present themselves. There were still the odd disconnect.

Also upon disconnecting and then re-establishing the publishers (`/fmu/out/*`) do not seem to restore and you are only left with the `/fmu/in/*` topics.

russkel, 2023-02-27:
> Increasing the packet loss counter in the ping check, fixed the problem.
>
> I set it to 5. [...]

I have increased to 5 as well and I am still getting ping timeouts and then disconnects. @dagar were you able to test this on hardware of yours?

biagio-trimarchi, 2023-09-26:
Hi,
I think I am facing the same problem on my setup.

I was trying to understand why the uxrce_dds_client disconnect from ROS2 and then immediately reconnect and I found this topic.

I have a pxhawk 6C connected to the RX/TX pins of a Jeston Xavier NX and the microdds connection randomly disconnect for a fraction of second while running, and then immediately reconnect.

I will try to change the num_ping_missed threshold as done in this topic and I will let you know.

Have anyone found any other solution in the mean time?