PX4/PX4-Autopilot #20854: Run multi uav depth camera

Cavalletta98, 2023-01-02:
Hi, i need some help to launch multiple uav with depth camera. I'm trying to use the [avoidance software ](https://github.com/PX4/PX4-Avoidance) with no success. In particular i cannot differenciate the frames of the different uav in order to use the `static_transform_publisher`
ALso, i noticed, that the second drone is not working. I mean, i cannot see it inb QGControl
Here is the launch file that i'm using
[goap_avoid.txt](https://github.com/PX4/PX4-Autopilot/files/10331075/goap_avoid.txt)

nicolasfripp, 2023-10-30:
Hey @Cavalletta98 . Did you find a solution for that? I'm facing the same issue

Cavalletta98, 2023-10-31:
Hi @nicolasfripp , i'm not workling anymore on this because the avoidance software has a lot of problems with multiple drones. It should be modified a lot in order to work

NishantChandna1403, 2025-04-25:
Hi, Could you a solution to this problem? I'm facing the same issue