PX4/PX4-Autopilot #20862: External Odometry Source

bilalkah, 2023-01-04:
Hi,
I am trying to feed PX4 Autopilot with external slam algorithm. However, I could get barely successful flight indoor environment. I use same approach with [vio](https://github.com/Auterion/VIO.git). I forward odometry messages through /mavros/odometry/out topic. But autopilot having trouble to following it. Also I examine the output is more reliable when the system is disarm. When system armed, it seems acting weirdly. 
Also I tried to feed odometry through /mavros/vision_pose/vision but it is also not successful run... I realized when I dont gave speed to the autopilot, the flight experience was poor and there was toilet effect... So I did not understand where is the problem. Can you help me?

pinkfloydfan, 2023-01-05:
Have you verified the mounting orientation of your VIO device? Toilet bowling is a symptom of an incorrect transform between VIO frame to aircraft frame. Your VIO velocities should be NED in aircraft body frame and positions must be NED in world frame, and the yaw offset between your VIO device and autopilot must also be accounted for. QGroundControl lets you graph the components of the Odometry message and that’s a good place to start debugging.

bilalkah, 2023-01-05:
Correct me if I am wrong, I believe we feed /mavros/odometry/out with odometry messages with ENU frame because I see T265 camera is giving in that frame and mavros odom plugin converting it to NED frame and forward it to FCU. I did the same, my odometry source is LIO-SAM algorithm. Only difference is the speed vector frame, because LIO-SAM generating speed vector according to map frame. So I convert it to ENU frame according to base_link.

pinkfloydfan, 2023-01-05:
Correct, MAVROS takes them in ENU and does the NED transformation. However have you verified the MAVLink odometry message? MAVLink inspector in QGC should let you do that easily.

bilalkah, 2023-01-06:
Yes I can see the odometry message in QGC. But the problem is FCU have been struggling follow it. I dont know how to solve it. I want FCU to obey the odometry I gave. Did you have any problem like that? @pinkfloydfan

pinkfloydfan, 2023-01-06:
when you move the drone forward/right/down is vx, vy and vz in the Odometry message increasing? When I've encountered the behaviour you describe using external vision it's usually because x, y, z components of the Odometry message are in the wrong order, or if there's a wrong transform between the external vision device's body frame and the drone's body frame.

bilalkah, 2023-01-09:
Yes I can see correct changes in odometry message when I move the vehicle. Does a continuous vibration in the odometry message negatively affect the autopilot?