PX4/PX4-Autopilot #20875: error when running roslaunch px4 posix_sitl.launch

Daidalos92, 2023-01-06:
Hi, I encounted an issue when trying to start the gazebo sitl with ROS. When trying to launch the gazebo simulation, I encountered the following error. 
```

**user@user:~/src/Firmware$ roslaunch px4 posix_sitl.launch
**WARNING: Package name "cppPub" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
multiple files named [posix_sitl.launch] in package `[px4]:`
- /home/user/src/Firmware/launch/posix_sitl.launch
- /home/user/src/Firmware/Firmware/launch/posix_sitl.launch
Please specify full path instead
The traceback for the exception was written to the log file****

```
Did anyone encountered similar issues or have an idea on why I am having two sets of launch files? Thanks in advance and let me know if any further info would be helpful.

Jaeyoung-Lim, 2023-01-06:
@Daidalos92 You have cloned a copy of this repository inside the firmware directory

Daidalos92, 2023-01-06:
Hi Jaeyoung, thanks for your fast reply. I removed the extra Firmware folder and was able to launch the gazebo simulation and mavros using the command "roslaunch px4 mavros_posix_sitl.launch". However, when I want running without ROS to try taking off using commander takeoff, i encountered the following error on a missing "flow_opencv.cpp" file when doing the make command. Any ideas on what may have went wrong with my installation? Thanks! 

```
user@user:~/src/Firmware$ no_sim=1 make px4_sitl_default gazebo
ninja: Entering directory `/home/user/src/Firmware/build/px4_sitl_default'
[2/6] Performing configure step for 'sitl_gazebo'
FAILED: cd /home/user/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/user/src/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/user/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
-- install-prefix: /usr/local
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Boost version: 1.58.0
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
Gazebo version: 7.16
-- Using C++14 compiler
-- Configuring done
CMake Error at external/OpticalFlow/CMakeLists.txt:73 (add_library):
  Cannot find source file:

    src/flow_opencv.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

-- Generating done
-- Build files have been written to: /home/user/src/Firmware/build/px4_sitl_default/build_gazebo
ninja: build stopped: subcommand failed.
Makefile:171: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

```

Jaeyoung-Lim, 2023-01-06:
@Daidalos92 Have you checked whether that file exists as shown in the log? It looks like you have not recursively updated the submodule

Daidalos92, 2023-01-06:
Hi Jaeyoung, have checked the src folder in my home directory and there is no opencv.cpp file but just the Firmware folder. May I know what you refer to by not recursively updating the submodule? thanks!

Daidalos92, 2023-01-10:
I am following this tutorial for offboard using python from the px4 documentation. Using ubuntu kinetic. When excuting roslaunch, I encountered the following error where it states its unable to launch [offb_node_py-7] and it auto shutsdown the program. Also i noticed that it escaltes to SIGTERM when it shutsdown. would that be a cause of concern? thanks in advance for your guidance 

```
user@user:~/src/Firmware$ roslaunch offboard_py start_offb.launch
WARNING: Package name "cppPub" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
... logging to /home/user/.ros/log/7cf1f614-9093-11ed-8abf-f45c89c15fb5/roslaunch-user.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "cppPub" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
started roslaunch server http://user:46546/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
[... 125 lines cut ...]
    from std_msgs.msg import Header
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/std_msgs/msg/_Bool.py", line 5, in <module>
    import genpy
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genpy/message.py", line 46, in <module>
    import genmsg
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/__init__.py", line 33, in <module>
    from . base import MSG_DIR, SRV_DIR, EXT_MSG, EXT_SRV, SEP, log, plog, InvalidMsgSpec, log_verbose, MsgGenerationException
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/genmsg/base.py", line 52, in <module>
[gazebo_gui-4] killing on exit
    import inspect, pprint
  File "/usr/lib/python2.7/inspect.py", line 41, in <module>
    from operator import attrgetter
KeyboardInterrupt
[gazebo-3] killing on exit
[sitl-2] killing on exit
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
ERROR [px4] Startup script returned with return value: 2
pxh> [sitl-2] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
```

Daidalos92, 2023-01-10:
link to px4 offboard documentation 

https://docs.px4.io/main/en/ros/mavros_offboard_python.html

closed 2023-01-16 by Daidalos92