PX4/PX4-Autopilot #20882: px4_fmu-v5 ostest priority_inheritance hard fault

dagar, 2023-01-07:
On px4_fmu-v5_test (px4_fmu-v5_default + test modules, nuttx ostest, etc) there's a hard fault when running the NuttX ostest.

[image]

Examples
 - http://px4-jenkins.dagar.ca:8080/blue/organizations/jenkins/PX4-Autopilot/detail/PR-20880/3/pipeline

dagar, 2023-01-07:
The ostest is failing intermittently on px4_fmu-v5_test (maybe 10-20% of the runs), but seems to be solid on the other architectures (stm32f4, stm32h7, kinetis).

dagar, 2023-01-07:
http://px4-jenkins.dagar.ca:8080/blue/organizations/jenkins/PX4-Autopilot/detail/pr-boards_fmu-v6x_ext2_fix/1/pipeline/530

dagar, 2023-01-07:
http://px4-jenkins.dagar.ca:8080/blue/organizations/jenkins/PX4-Autopilot/detail/pr-simulation_airframes_contd/7/pipeline/454

davids5, 2023-01-08:
@dagar i am Oot till 17th.

ThomasDebrunner, 2023-01-11:
Can't reproduce it on my fmu_v5 with (building `px4_fmu-v5_test ` from main, running ostest). Will test some more

dagar, 2023-01-11:
> Can't reproduce it on my fmu_v5 with (building `px4_fmu-v5_test ` from main, running ostest). Will test some more

Which FMUv5 variant and is the airframe configured? Might just be a coincidence, but so far all of the `ostest` failures are happening on either the CUAV v5plus (mostly) and Holybro Pixhawk 4. There are 4 variants of the FMUv5 on the hardware test rack (Holybro Pixhawk 4, 4 mini with no px4io, CUAV v5 plus, CUAV v5 nano with no px4io) and Jenkins will schedule any of them to run these jobs.

dagar, 2023-01-14:
Anecdata, I don't think I've seen this failure on the test rack since bumping the interrupt stack last week. https://github.com/PX4/PX4-Autopilot/commit/dc5ce9b0cecc1a165af0b639a903ec1d694e0e80