PX4/PX4-Autopilot #20892: ninja: error: unknown target 'gazebo', did you mean 'geo'? make: *** [Makefile:227: px4_sitl_default] Error 1

98uranus, 2023-01-09:
I am running Ubuntu Linux 20.04 on Parallels, attempting to build a PX4 HITL simulation. 

I then installed the PX4 autopilot by running 'git clone https://github.com/PX4/PX4-Autopilot.git --recursive' and then 
'bash ./PX4/Autopilot/Tools/setup/ubuntu.sh'

This completed but with an error and some warnings:

-- PX4 version: v1.14.0-beta1-168-gaf336719e6
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/uranus/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
-- Looking for ignition-transport8 -- found version 8.4.0
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Checking for module 'uuid'
--   Found uuid, version 2.34.0
-- Found UUID: TRUE  
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.13.0
-- Searching for dependencies of ignition-math6
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.2.0
-- Found Java: /home/uranus/.sdkman/candidates/java/current/bin/java (found version "1.8.0_322") 
-- ROMFS: ROMFS/px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/uranus/PX4-Autopilot/build/px4_sitl_default
[0/2] Re-checking globbed directories...
ninja: error: unknown target 'gazebo', did you mean 'geo'?
make: *** [Makefile:227: px4_sitl_default] Error 1

please help!!

RadioFront, 2023-01-09:
I guess you faced this error during build `make px4_sitl gazebo`. I get the exactly same error in MacOS.
`[0/2] Re-checking globbed directories...`
`ninja: error: unknown target 'gazebo', did you mean 'geo'?`
`make: *** [px4_sitl] Error 1`

98uranus, 2023-01-09:
ㅛes, this error occured during build make px4_sitl gazebo. This error did not occur when built with an old autopilot file, but it occurred with a newly downloaded file.

RadioFront, 2023-01-09:
A new github download (6:27 UTC 1/9/23) causes error with `jmavsim` too
`make px4_sitl jmavsim`
`[0/2] Re-checking globbed directories...`
`ninja: error: unknown target 'jmavsim'`

95ImadEL, 2023-01-09:
> I get the exactly same error in MacOS.

I get the same error in Ubuntu 20.04.

beniaminopozzan, 2023-01-09:
regarding `make px4_sitl gazebo`, since #20867 it is necessary to specify the model
So now it is `make px4_sitl gazebo_iris`

98uranus, 2023-01-11:
Thank you for your reply

EG-Julien, 2023-01-12:
If you're still facing some problems you could checkout to this version where it still works [bb591c6](https://github.com/PX4/PX4-Autopilot/commit/bb591c6c29b49622bf79f2528b8382c930f3720e).

naifaljohani, 2023-03-09:
Here is how I solved it : 
1- After installing WSL ,I used the installed Ubuntu 22.04, if that is the case , you need to install Gazebo manually in Ubuntu.
2- Then you can run :
make px4_sitl gazebo

If you want more options, not only a Quad,  you need to refert to ubuntu 20.04, and install  Gazebo Classic , for more details see the link :
http://docs.px4.io/main/en/simulation/

naifaljohani, 2023-03-09:
Here is how I solved it : 
1- After installing WSL ,I deleted the installed Ubuntu , then reinstall Ubuntu 22.04 ,then install Gazebo manually in Ubuntu following stepes in Gazebo wbsite.
2- Then you can run :
make px4_sitl gazebo

If you want more options, not only a Quad,  you need to refert to ubuntu 20.04, and install  Gazebo Classic , for more details see the link :
http://docs.px4.io/main/en/simulation/

Inglam, 2024-01-12:
> A new github download (6:27 UTC 1/9/23) causes error with `jmavsim` too `make px4_sitl jmavsim` `[0/2] Re-checking globbed directories...` `ninja: error: unknown target 'jmavsim'`

Exactly same here. Have you found a solution yet?

Comor543, 2024-07-22:
> regarding `make px4_sitl gazebo`, since #20867 it is necessary to specify the model So now it is `make px4_sitl gazebo_iris`

Didn't work for me...
Still get the same error.

Gson-glitch, 2024-08-20:
For those experiencing this issue Ubuntu 22.04 and using gazebo harmonic or garden, try this command instead
`make px4_sitl gz_x500`
[PX4 Gazebo Simulation](https://docs.px4.io/main/en/sim_gazebo_gz/)

Meowchine1, 2025-02-28:
> Для тех, кто столкнулся с этой проблемой в Ubuntu 22.04 и использует Gazebo harmonic или garden, попробуйте эту команду вместо `make px4_sitl gz_x500` [PX4 Gazebo Simulation](https://docs.px4.io/main/en/sim_gazebo_gz/)
I have ```gz_x500``` in ```/PX4-Autopilot/ROMFS/px4fmu_common/init.d-posix/airframes/ ``` but I got 
```[Err] [UserCommands.cc:1152] Error Code 1: Msg: Error finding file [x500/model.sdf].
[Err] [UserCommands.cc:1152] Error Code 1: Msg: Unable to read file: [x500/model.sdf]
WARN  [gz_bridge] Service call timed out: /gui/follow
WARN  [gz_bridge] Service call timed out: /gui/follow/offset
ERROR [gz_bridge] timed out waiting for clock message
ERROR [gz_bridge] Task start failed (-1)
ERROR [init] gz_bridge failed to start and spawn model
ERROR [px4] Startup script returned with return value: 256
FAILED: src/modules/simulation/gz_bridge/CMakeFiles/gz_x500 /home/vor/PX4-Autopilot/build/px4_sitl_default/src/modules/simulation/gz_bridge/CMakeFiles/gz_x500 
cd /home/vor/PX4-Autopilot/build/px4_sitl_default/src/modules/simulation/gz_bridge && /usr/bin/cmake -E env PX4_SIM_MODEL=gz_x500 /home/vor/PX4-Autopilot/build/px4_sitl_default/bin/px4
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1```

closed 2025-02-28 by dakejahl

cimone123, 2025-05-21:
> For those experiencing this issue Ubuntu 22.04 and using gazebo harmonic or garden, try this command instead对于那些遇到这个问题并且使用 Ubuntu 22.04 以及 gazebo harmonic 或 garden 的用户，尝试这个命令 `make px4_sitl gz_x500` [PX4 Gazebo Simulation  PX4 Gazebo 仿真](https://docs.px4.io/main/en/sim_gazebo_gz/)

Can Gazebo Harmonic or garden also be used in HITL simulation?

Gson-glitch, 2025-05-21:
@cimone123 Unfortunately, no. As of now, HITL simulation is not officially supported in Gazebo Harmonic or Gazebo Garden within the PX4 ecosystem. The current HITL support is primarily available through Gazebo Classic and jMAVSim simulators. 
https://github.com/PX4/PX4-Autopilot/issues/22431

Louis-fortress, 2025-12-29:
Hello guys please I face this issue when I try to load my custom models and worlds in gazebo. I place my custom models in the model directory for simulation, create the airframe for my px4_sitl, place it in the cmakelist but it still doesn't load.