PX4/PX4-Autopilot #20914: Weird EKF behaviour with VIO

ripdk12, 2023-01-12:
So I'm using a VIO source which provides very accurate odometry messages sent to the px4 via. MAVROS. It does fly well, but occasionally the state estimator flips out completely which has the potential to lead to some pretty nasty crashes.

A flight log of this happening: https://review.px4.io/plot_app?log=b3cfc2fb-83ef-431a-890c-e9b905b5fb6f

I've been experimenting with all kinds of values in the parameter list, like the EKF2_AID_MASK, but the issue prevails.
I've noticed when looking at my rosbag, that the VIO data i feed is always correct, and the "local_position/odom" message, which i assume to be after EKF2 fusion, is never quite as precise, and sometimes completely trusts the GPS position (which is very inaccurate).

How can I tune the EKF to trust the external vision system more than it currently does?

dagar, 2023-01-13:
Could you give this a try with current PX4 main and share the log? There's been quite a lot of churn in this area and we're now in a position to better position to work through it in main (soon to be v1.14).

ripdk12, 2023-01-13:
> Could you give this a try with current PX4 main and share the log? There's been quite a lot of churn in this area and we're now in a position to better position to work through it in main (soon to be v1.14).#20304

I would love to try in main, but I'm too much of a noob to figure out how to flash main onto my cube orange plus.
Trying to implement the changes from #20304 i get this error when trying to build with:
 ```make cubepilot_cubeorangeplus_default```

```
-- Found PythonInterp: /usr/bin/python3 (found version "3.10.6") 
-- Using C++11
-- Release build type: MinSizeRel
-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding platforms/nuttx/init/stm32h7/rc.board_arch_defaults -> /etc/init.d/rc.board_arch_defaults
-- ROMFS:  Adding boards/cubepilot/cubeorangeplus/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- ROMFS:  Adding boards/cubepilot/cubeorangeplus/init/rc.board_sensors -> /etc/init.d/rc.board_sensors
-- ROMFS:  Adding boards/cubepilot/cubeorangeplus/init/rc.board_mavlink -> /etc/init.d/rc.board_mavlink
-- ROMFS:  Adding boards/cubepilot/cubeorangeplus/extras/cubepilot_cubeorangeplus_bootloader.bin -> /etc/extras/cubepilot_cubeorangeplus_bootloader.bin
-- ROMFS:  Adding boards/cubepilot/cubeorangeplus/extras/cubepilot_io-v2_default.bin -> /etc/extras/cubepilot_io-v2_default.bin
CMake Error at platforms/nuttx/CMakeLists.txt:333 (message):
  init scripts require NSH SET (board has CONFIG_NSH_DISABLE_ECHO=y)

-- Configuring incomplete, errors occurred!
See also "/home/hecto/PX4-Autopilot/build/cubepilot_cubeorangeplus_default/CMakeFiles/CMakeOutput.log".
See also "/home/hecto/PX4-Autopilot/build/cubepilot_cubeorangeplus_default/CMakeFiles/CMakeError.log".
Error: /home/hecto/PX4-Autopilot/build/cubepilot_cubeorangeplus_default is not a directory
make: *** [Makefile:232: cubepilot_cubeorangeplus] Error 1
```

Is this something you could help me with?

If it's easier we could also move this to discord?
Thanks!

ripdk12, 2023-01-16:
So I was able to flash main to cube orange plus now (with #20304) and have tested here:

https://review.px4.io/plot_app?log=9fd1eac0-435e-49f6-bb46-05f5b67ff766

The drone seems to be flying better now, but I still get this random funny behavior every once in a while.
I'm literally just telling it to fly in a 1x1 meter square and land at starting position.