PX4/PX4-Autopilot #20918: px4_fmu-v4_default target fails due to cmath change

marcojob, 2023-01-12:
## Describe the bug
This [PR](https://github.com/PX4/PX4-Autopilot/pull/19893/files) seems to be making the `px4_fmu-v4_default` target fail for me (no matter the cross compiler version).

The error is the following:
```
PX4-Autopilot/platforms/nuttx/NuttX/include/cxx/cmath:60:13: error: conflicting declaration of C function 'bool std::isfinite(double)'
 inline bool isfinite(double value) { return __builtin_isfinite(value); }
```

Running the `make px4_fmu-v6x_default` target works on the other hand, so that's I think its something specific to the v4 target. Couldn't figure out so far what exactly could be wrong.

dagar, 2023-01-13:
It's still working for me locally and in CI, what's your build environment exactly? Maybe share the initial cmake configure output from a completely clean state?

marcojob, 2023-01-13:
@dagar thanks for your comment. I must have a mistake yesterday evening selecting the right compiler as the target works on my laptop with compiler version 9.3.1. It does not work with 7.2.1 (which is fine).

closed 2023-01-13 by marcojob