PX4/PX4-Autopilot #20927: make px4_sitl gazebo_typhoon_h480 fails after mavlink git version check

Smashroom, 2023-01-13:
Heya,

I am trying to create a docker image that will let me build a ROS2+Gazebo+PX4 environment with DDS support.

**Environment**
Docker image: px4io/px4-dev-ros2-galactic:latest
PX4 commit: 1e93ae31482f6bdc64bc8120f5f2e5508dfef625
Gazebo installed via: `curl -sSL http://get.gazebosim.org | sh` 
MicroDDS agent installed via: `https://github.com/micro-ROS/micro_ros_setup.git`
Python version: 3.8.10

**To run the environment**
```
docker run -it --privileged\
    --rm \
    --net=host \
    --env="DISPLAY=$DISPLAY" \
    --env="QT_X11_NO_MITSHM=1" \
    -v src/Firmware:/src/firmware/:rw \
    --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
    --env="XAUTHORITY=$XAUTH" \
    --volume="$XAUTH:$XAUTH" \
    --runtime=nvidia \
     image_name bash
```

**Error**:
```
cd /src/firmware && /usr/bin/python3 /src/firmware/src/lib/version/px_update_git_header.py /src/firmware/build/px4_sitl_default/src/lib/version/build_git_version.h --validate
Traceback (most recent call last):
  File "/src/firmware/src/lib/version/px_update_git_header.py", line 139, in <module>
    mavlink_git_version = subprocess.check_output('git rev-parse --verify HEAD'.split(),
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'rev-parse', '--verify', 'HEAD']' returned non-zero exit status 128.
[159/878] Building CXX object msg/CMakeF...r/topics_sources/sensor_correction.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1
```

*build_git_version.h* is not present:

```
root@xxx:/# cat /src/firmware/build/px4_sitl_default/src/lib/version/build_git_version.h
cat: /src/firmware/build/px4_sitl_default/src/lib/version/build_git_version.h: No such file or directory
```

I couldn't get rid of the error whatever I tried. I didn't figure out a way to generate `build_git_version.h`. Did anyone faced that with the recent commits?

AlexWUrobot, 2023-01-19:
I have a similar issue and am still solving the issue.
[image]

AlexWUrobot, 2023-01-20:
I guess one reason is the newest version PX4-Autopilot change the setup_gazebo.bash location.
Thus we have to change the bashrc file like this :
# source ~/px-ws/src/PX4-Autopilot/Tools/setup_gazebo.bash ~/px-ws/src/PX4-Autopilot ~/px-ws/src/PX4-Autopilot/build/px4_sitl_default
source ~/px-ws/src/PX4-Autopilot/Tools/simulation/gazebo/setup_gazebo.bash ~/px-ws/src/PX4-Autopilot ~/px-ws/src/PX4-Autopilot/build/px4_sitl_default

[image]

ChanJoon, 2024-01-29:
I have same issue on docker image `px4io/px4-dev-ros-noetic`. Without mounting local PX4 src, I cloned repository in docker `/home/user`. After doing some installation steps, `make px4_sitl gazebo-classic` command shows following errors.

```bash
FAILED: src/lib/version/build_git_version.h 
cd /home/user/PX4-Autopilot && /usr/bin/python3 /home/user/PX4-Autopilot/src/lib/version/px_update_git_header.py /home/user/PX4-Autopilot/build/px4_sitl_default/src/lib/version/build_git_version.h --validate
Traceback (most recent call last):
  File "/home/user/PX4-Autopilot/src/lib/version/px_update_git_header.py", line 139, in <module>
    mavlink_git_version = subprocess.check_output('git rev-parse --verify HEAD'.split(),
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'rev-parse', '--verify', 'HEAD']' returned non-zero exit status 128.
```

ChanJoon, 2024-01-29:
I think I found the reason.
Previously, I installed PX4-Autopilot at `/home/user`. Maybe `px4_update_git_header.py` requires some certain directory.

After I removed the source code and re-installed at `~/`, I successfully build with `make px4_sitl gazebo-classic`. It might not be correct reason but I didn't do any other stuff.

closed 2025-10-26 by github-actions[bot]