PX4/PX4-Autopilot #20929: Pre flight failure with Optical Flow on Ground

AlexKlimaj, 2023-01-14:
In current main, with optical flow fusion enabled, I am unable to arm in altitude or position mode. If I take off in manual I am able to switch to altitude and position no problem. 

On a larger quad with long landing gear, this issue doesn't present itself. It appears to be a problem with the optical flow close to the ground.

AlexKlimaj, 2023-01-15:
https://github.com/PX4/PX4-Autopilot/pull/20940

AlexKlimaj, 2023-01-24:
It looks like this is still an issue that was introduced with this commit.
https://github.com/PX4/PX4-Autopilot/pull/20414

https://github.com/PX4/PX4-Autopilot/pull/20414/files#diff-a40788d784ca3877b825cbf025e2f84b154faf5182ae81a27defa625f27cffa3R943
Starts off with real values then goes INF pretty quickly.

```
 I0NFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00006 vel: 0.00006
0I.NFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00006 vel: 0.00006
.I0NFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00006 vel: 0.00006
0I0NFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00006 vel: 0.00006
0I0NFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00006 vel: 0.00006
0I0NFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00006 vel: 0.00006
0I0NFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00006 vel: 0.00006
0I      NFO  [ekf2] test_ratio_max: 0.00000     of_vel: 0.00006 vel: 0.00006
        IoNFO  [ekf2] test_ratio_max: 0.00000   of_vel: 0.00006 vel: 0.00006
oIfNFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00006 vel: 0.00006
fI_NFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00006 vel: 0.00006
_IvNFO  [ekf2] test_ratio_max: 0.00308  of_vel: 0.05552 vel: 0.05552
vIeNIFO  [ekf2] test_ratio_max: 0.00003 of_vel: 0.00563 vel: 0.00563
eIlNNFO  [ekf2] test_ratio_max: 0.00003 of_vel: 0.00563 vel: 0.00563
lI:NFFO  [ekf2] test_ratio_max: 0.00006 of_vel: 0.00783 vel: 0.00783
:I NOFO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
 I0N FO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
0.. 00004       vel: 0.00004
I0N[FO  [ekf2] test_ratio_max: inf      of_vel: inf     vel: inf
0IuNFO  [ekf2] test_ratio_max: 0.00000  of_vel: 0.00004 vel: 0.00004
I0NaFO  [ekf2] test_ratio_max: inf      of_vel: inf     vel: inf
II0NvFO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
WARN  [health_and_arming_checks] Preflight Fail: velocity estimate error
N0FcO  [ekf2] test_ratioINFO  [ekf2] test_ratio_max: inf        of_vel: inf     vel: inf
_I      NaFO  [ekf2] test_ratio_max: inf        of_vel: inf     vel: inf
mIvNnFO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
aex:: 0.00000   of_vel: 0.00004 vel: 0.00004
IlN1FO  [ekf2] test_ratio_max: inf      of_vel: inf     vel: inf
II:N2FO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
N F5O  [ekf2] test_ratio_max: 0.00000   of_vel: 0.00004 vel: 0.00004
I0N:FO  [ekf2] test_ratio_max: inf      of_vel: inf     vel: inf
II.NuFO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
N0FaO  [ekf2]INFO  [ekf2] test_ratio_max: inf   of_vel: inf     vel: inf
 I0NvFO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
tI0NcFO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
e0sat_ratio_max: 0.00000        of_vel: 0.00004 vel: 0.00004
I0NnFO  [ekf2] test_ratio_max: inf      of_vel: inf     vel: inf
II
NnFO  [ekf2] test_ratio_max: inf        of_vel: inf     vel: inf
NIFNoFO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
INFO  [ekf2] test_ratio_max: inf        of_vel: inf     vel: inf
OIINdFO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
 INNeFO  [ekf2] test_ratio_max: inf     of_vel: inf     vel: inf
```

dirksavage88, 2023-05-08:
commit hash: 7eb6535273b887ae31887d45b12221c5320d9bda

Pretty much the same the optical flow issues in #21055 

Is there a way around this for the time being, or just wait for the fix? For reference I'm using the Bitcraze flowdeck v2 (pmw3901) over SPI bus. I am able to see the optical flow sensor messages and have a vl53l1x distance sensor. Tried raising the drone 1m off the ground as you had done in #21358 but still getting constant spamming of velocity estimate errors in the system console.

junwoo091400, 2023-05-10:
> Is there a way around this for the time being, or just wait for the fix?

During a [community qna call](https://discuss.px4.io/t/px4-community-q-a-may-03-2023/31965#d4-release-candidate-for-v114-15), @dagar hinted that we can have a quick safeguard fix to get around this issue. Could you comment on this?

AlexKlimaj, 2023-06-27:
https://github.com/PX4/PX4-Autopilot/pull/21769

bresch, 2023-07-20:
Fixes merged

closed 2023-07-20 by bresch

xiedimeng889, 2025-04-28:
My simulation model has turned off GPS, optical flow and distance sensor are effective, and can take off using position mode, but there is a 10% chance of takeoff failure,
```
param set-default EKF2_OF_CTRL 1
param set-default EKF2_TERR_MASK 3
param set-default MIS_TAKEOFF_ALT 2
param set-default RTL_DESCEND_ALT 3
param set-default EKF2_BARO_CTRL 1
param set-default EKF2_HGT_REF 2 
param set-default EKF2_RNG_CTRL 1
```
The code version I am using is v1.14.3