PX4/PX4-Autopilot #20931: Build problem with `make px4_sitl gazebo`

ba-13, 2023-01-14:
## Describe the bug

Facing the following build error while trying `make px4_sitl gazebo`.

## Expected behavior

`make px4_sitl gazebo` builds successfully.

## Logs

```bash
➜  PX4-Autopilot git:(main) make px4_sitl gazebo
[0/2] Re-checking globbed directories...
[0/4] Performing build step for 'sitl_gazebo'
[1/17] Generating include/moc_gazebo_user_camera_plugin.cpp
FAILED: include/moc_gazebo_user_camera_plugin.cpp 
cd /home/ba13/PX4-Autopilot/build/px4_sitl_default/build_gazebo/include && /usr/lib/qt5/bin/moc @/home/ba13/PX4-Autopilot/build/px4_sitl_default/build_gazebo/include/moc_gazebo_user_camera_plugin.cpp_parameters
usr/include/cstrin:73: Parse error at "std"
[2/17] Generating include/moc_gazebo_video_stream_widget.cpp
FAILED: include/moc_gazebo_video_stream_widget.cpp 
cd /home/ba13/PX4-Autopilot/build/px4_sitl_default/build_gazebo/include && /usr/lib/qt5/bin/moc @/home/ba13/PX4-Autopilot/build/px4_sitl_default/build_gazebo/include/moc_gazebo_video_stream_widget.cpp_parameters
usr/include/cstrin:73: Parse error at "std"
[3/17] Linking CXX shared library libLiftDragPlugin.so
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/ba13/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/local/bin/cmake --build /home/ba13/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 3
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1
```

dagar, 2023-01-14:
Thanks for reporting, could you please try https://github.com/PX4/PX4-Autopilot/pull/20934?

dagar, 2023-01-14:
What's your dev environment and px4 version? Did you ever have this working?

ba-13, 2023-01-15:
I tried with the PR you mentioned above, it gave _nearly_ the same error
```bash
Copying fixed headers for protocol 2.0 to /attached/Aerial_Robotics/PX4/build/px4_sitl_default/mavlink
[9/810] Performing build step for 'sitl_gazebo'
[2/26] Generating include/moc_gazebo_video_stream_widget.cpp
FAILED: include/moc_gazebo_video_stream_widget.cpp 
cd /attached/Aerial_Robotics/PX4/build/px4_sitl_default/build_gazebo/include && /usr/lib/qt5/bin/moc @/attached/Aerial_Robotics/PX4/build/px4_sitl_default/build_gazebo/include/moc_gazebo_video_stream_widget.cpp_parameters
usr/include/cstrin:73: Parse error at "std"
[6/26] Building CXX object CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o
ninja: build stopped: subcommand failed.
[806/810] Linking CXX executable bin/px4
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /attached/Aerial_Robotics/PX4/build/px4_sitl_default/build_gazebo && /usr/local/bin/cmake --build /attached/Aerial_Robotics/PX4/build/px4_sitl_default/build_gazebo -- -j 5
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1
```
Noting that `FAILED: include/moc_gazebo_user_camera_plugin.cpp` no longer fails this time, or at least it's not showing up.

> What's your dev environment and px4 version? Did you ever have this working?

I'm not sure what you want to know by dev environment. I'm currently using Ubuntu 20.04. Can you please explain what info you need? Also regarding px4 version, I'm just using the source code, main branch.

marceloroberti, 2023-01-17:
#20775

marceloroberti, 2023-01-17:
I had a problem like this when I first tried to build gazebo simulation.
Try these solutions that I received in my issue: #20775