PX4/PX4-Autopilot #20932: Jmavsim Build fails: no target

ecmnet, 2023-01-14:
Latest master fails to build using the new build notation:

`make px4_sitl jmavsim_iris`

as a result I get target jmavsim_Iris not found.

dagar, 2023-01-14:
The targets will only exist if the dependencies are found. https://github.com/PX4/PX4-Autopilot/blob/82ecbccd85015e36c84ea54904b3f32144bee282/src/modules/simulation/simulator_mavlink/sitl_targets_jmavsim.cmake#L2-L13

What's your environment exactly? From a clean state can you share the output of `make px4_sitl_default jmavsim_iris`? I'm looking for the cmake configure output.

ecmnet, 2023-01-15:
Thanks for pointing me into the right direction. I did no see the issue as the message not finding vecmath.jar  currently is commented out

[image]

`vecmath.jar` however is installed in

`PX4-Autopilot/Tools/simulation/jmavsim/jMAVSim/lib`

but obviously not found. For testing, I removed the condition for vecmath.jar in `site_targets_jmavsim.cmake`. The target is created properly then, PX4 compiles but jmavsim is not started because jmavsim_run.sh is expected in the wrong place

  ```
CAL_MAG0_PRIO: curr: -1 -> new: 50
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CAL_MAG1_PRIO: curr: -1 -> new: 50
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
INFO  [init] jMAVSim simulator
PX4-Autopilot/build/px4_sitl_default/rootfs/etc/init.d-posix/px4-rc.simulator: line 139: ./jmavsim_run.sh: No such file or directory
INFO  [simulator_mavlink] using TCP on remote host localhost port 4560
WARN  [simulator_mavlink] Please ensure port 4560 is not blocked by a firewall.
INFO  [simulator_mavlink] Resolved host 'localhost' to address: 127.0.0.1
INFO  [simulator_mavlink] Waiting for simulator to accept connection on TCP port 4560

```

julianoes, 2023-01-16:
> The targets will only exist if the dependencies are found. 

That's going to cause many frustrated users, isn't it?

closed 2023-01-25 by ecmnet