PX4/PX4-Autopilot #20938: Stablizing a new build

kurosh-z, 2023-01-15:
I built a quadcopter similar to Holybro’s QAV250 but with an NXP flight controller. I installed px4 firmware and calibrated sensors and ESCs. I have the following issues:

- Controller issue: I noticed after arming I can increase the motor speed accurately with the RC controller but I can’t decrease it anymore before actually taking off ([Video](https://www.youtube.com/watch?v=QxuveSixiO4)).

- After taking off in stabilize or manual mode, the drone goes to one side, and it’s really hard to control. It barely left the surface, while it already flies to the right side and sometimes a little backward. I didn’t increase the throttle because it’ll make it move to the side even faster and harder to control ([video](https://www.youtube.com/watch?v=ZDMb_A4u8Kk)).

Any suggestion on how I can fix these issues? In particular, I would like to know if creating a new custom airframe helps.

dagar, 2023-01-15:
Can you share the corresponding flight logs? https://logs.px4.io

kurosh-z, 2023-01-17:
Sorry for the delay. I couldn't find the log files for that flight for some reason, but today I managed to do some new videos and collect the log files. 
For the first experiment, I limited the drone movement with a cable [with_cable](https://youtu.be/3-1IBakPv7M) 
for the 2nd one, I removed the cable. as you can see the drone drifts toward the camera and backward. [no_trim](https://youtu.be/emHY5HuS-FI)
so I decided to fix this by adjusting sub-trims of  the RC controller [subtrim_01](https://youtu.be/E_jIMQYvUBA)
the next attempt  to fix the backward and right drifting with more sub-trim percentage:
[subtrim_02](https://youtu.be/wHlyC8jNjeo), [subtrim_03](https://youtu.be/uX2bBNQR6kA), [subtrim_04](https://youtu.be/mV8r2p9t7ow)
but I couldn't stabilize the drone in the end. 

log files:
https://logs.px4.io/plot_app?log=8b993afc-b621-41b1-b392-c04faf813f45
https://logs.px4.io/plot_app?log=d4f06101-9ba4-44f0-af89-f2f9af990fdf
https://logs.px4.io/plot_app?log=964e5a12-5fd6-45ed-9724-fac9d27150c6
https://logs.px4.io/plot_app?log=75c4d817-15b6-4b13-ae5f-6bf5270bf21f
https://logs.px4.io/plot_app?log=b559dd42-b672-4956-a27a-c0e5dbefcde4
https://logs.px4.io/plot_app?log=8b265b1f-da81-44e6-addf-96e3a083fd6f