PX4/PX4-Autopilot #20941: QMCL5883L Z-axis problem

alihaanc, 2023-01-15:
[image]

```
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or Discord (you can find an invite link on this project README).

 **_These are the main parts of the code I wrote for the QMCL5883L but the z-axis is very noisy, I don't understand why. is this normal_**

##void QMC5883L_Start(QMC5883L *dev)
{
	uint8_t tempRx;
	HAL_StatusTypeDef status;

	status = HAL_I2C_Mem_Read(dev->i2c1_handle, R_Device_address,R_Status, SIZE_OF_ADDRES, &tempRx, SIZE_OF_DATA, TIMEOUT);

	if((tempRx & Data_Ready)== 1 && (status == HAL_OK)){

	HAL_I2C_Mem_Read_DMA(dev->i2c1_handle, R_Device_address, R_Data_LSB_X , SIZE_OF_ADDRES, dev->rx_buffer, 6);

	dev->RDY_flag = 1;

	tempRx = 0;

	} Describe your preferred solution
A clear and concise description of what you want to happen.

## Describe possible alternatives
A clear and concise description of alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots for the feature request here.

```
```
void HAL_I2C_MemRxCpltCallback(I2C_HandleTypeDef *hi2c){

	if(hi2c->Instance ==hi2c1.Instance){

		Int_Process_handler(&dev);

		dev.RDY_flag = 0;
	}

	}
```

```
void Int_Process_handler(QMC5883L*dev)
{
      dev->QMCL_Axis_Raw[0] = (int16_t)( dev->rx_buffer[1] << 8| dev->rx_buffer[0] );
      dev->QMCL_Axis_Raw[1] = (int16_t)( dev->rx_buffer[3] << 8| dev->rx_buffer[2] );
      dev->QMCL_Axis_Raw[2] = (int16_t)( dev->rx_buffer[4] << 8| dev->rx_buffer[5] );

      dev->heading = atan2(dev->QMCL_Axis_Raw[1] , dev->QMCL_Axis_Raw[0] ) ;

}
```
## These are the main parts of the code I wrote for the QMCL5883L but the z-axis is very noisy, I don't understand why. is this normal

dagar, 2023-01-15:
This is for issues with PX4 autopilot firmware. Are you seeing this noisy data when running current versions of PX4?

alihaanc, 2023-01-15:
> This is for issues with PX4 autopilot firmware. Are you seeing this noisy data when running current versions of  

>No ,I dont use PX. İt is stm32f411ceu6 (blackpill board)

alihaanc, 2023-01-15:
> > This is for issues with PX4 autopilot firmware. Are you seeing this noisy data when running current versions of  
> 
> 
> >No ,I dont use PX. İt is stm32f411ceu6 (blackpill board)
> this my own driver
>

closed 2023-01-15 by alihaanc

dagar, 2023-01-16:
You could take a look at our qmc5883l driver for reference, although there's not actually a lot to it. https://github.com/PX4/PX4-Autopilot/blob/main/src/drivers/magnetometer/qmc5883l/QMC5883L.cpp

I could see any settings that apply to the z axis specifically, but you could check the oversampling and range settings. https://github.com/PX4/PX4-Autopilot/blob/00bfd5436a6f743fe79d6c300ab37982fcf2987f/src/drivers/magnetometer/qmc5883l/QMC5883L.hpp#L109-L111