PX4/PX4-Autopilot #20942: "Invalid <arg> tag: mavlink_sitl_gazebo" when exceuting roslaunch px4 posix_sitl.launch

Daidalos92, 2023-01-16:
I encountered the following error when running the `roslaunch px4 posix_sitl.launch` command from the PX4 SITL tutorial. I am running on ubuntu 16.04 ROS kinetic. anyone encountered similar errors or know any possible causes for it? i have already sourced and exported the paths as per below. I am able to run the `no_sim=1 make px4_sitl_default gazebo` as per normal.

```
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/simulation/gazebo/sitl_gazebo
```

```
offboard@shawnmlee-MacBookPro:~/src/Firmware$ **roslaunch px4 posix_sitl.launch**
... logging to /home/offboard/.ros/log/2bc60f36-954e-11ed-830e-f45c89c15fb5/roslaunch-shawnmlee-MacBookPro-4132.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

**Invalid <arg> tag: mavlink_sitl_gazebo**
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/offboard/catkin_ws/src/mavlink
ROS path [2]=/home/offboard/catkin_ws/src/mavros/libmavconn
ROS path [3]=/home/offboard/catkin_ws/src/mavros/mavros_msgs
ROS path [4]=/home/offboard/catkin_ws/src/mavros/mavros
ROS path [5]=/home/offboard/catkin_ws/src/mavros/mavros_extras
ROS path [6]=/home/offboard/catkin_ws/src/mavros/test_mavros
ROS path [7]=/opt/ros/kinetic/share
ROS path [8]=/home/offboard/src/Firmware
ROS path [9]=/home/offboard/src/Firmware/Tools/simulation/gazebo/sitl_gazebo. 

Arg xml is <arg default="$(find mavlink_sitl_gazebo)/worlds/empty.world" name="world"/>
The traceback for the exception was written to the log file
```

Jaeyoung-Lim, 2023-01-16:
Which version of px4 are you running?

Daidalos92, 2023-01-16:
I git cloned the v1.9.2 tag from the main PX4 repository.

Jaeyoung-Lim, 2023-01-16:
@Daidalos92 I no longer remember how it was done in v1.9.2

You can check whether px4 is on the `ROS_PACKAGE_PATH`

Otherwise, Ubuntu 16.04 is is EOL and I recommend you to update your setup

Jaeyoung-Lim, 2023-01-16:
No you should have access to gazebo_ros through your package manager.

However, since Ubuntu 16.04 is end of life, I am not sure whether this is still available