PX4/PX4-Autopilot #20952: Cannot run the micrortps_agent

fbenti, 2023-01-17:
## Describe the bug
Ubuntu 20.04 LTS
ros2 foxy

After having installed the Fast-RTPS-Gen and the px4_ros_com (running the _build_ros2_workspace.bash_), I source the env with `
`source ~/px4_ros_com_ros2/install/setup.bash`

and the run:
`micrortps_agent -t UDP
`

but I get:
`micrortps_agent: command not found
`

What am I doing wrong?

ripdk12, 2023-01-18:
What version of the firmware are you using? If you're on main, this doesn't work, the RTPS bridge was discontinued, and the documentation is horribly outdated.

russkel, 2023-01-24:
You do not need to use `micrortps`. I am using `micro_ros_agent` from the vulcanexus distribution. That's an easy way to get up and running with it.

hendjoshsr71, 2023-01-25:
You may want to see some newer install instructions here, 

https://gist.github.com/julianoes/adbf76408663829cd9aed8d14c88fa29

I'd imagine you will need `humble` with `foxy`.