PX4/PX4-Autopilot #20966: why the getEffectivenessMatrix of ActuatorEffectivenessMCTilt didn't update the aixs of rotor from tilt?

mengchaoheng, 2023-01-19:
we can see that the [ActuatorEffectivenessMCTilt::getEffectivenessMatrix](https://github.com/PX4/PX4-Autopilot/blob/5eb13e44480ab4f896c253b77a525f5286ded84a/src/modules/control_allocator/ActuatorEffectiveness/ActuatorEffectivenessMCTilt.cpp#L46) function didn't use function like [ActuatorEffectivenessRotors::updateAxisFromTilts](https://github.com/PX4/PX4-Autopilot/blob/5eb13e44480ab4f896c253b77a525f5286ded84a/src/modules/control_allocator/ActuatorEffectiveness/ActuatorEffectivenessRotors.cpp#L236) which is used in [ActuatorEffectivenessTiltrotorVTOL::getEffectivenessMatrix](https://github.com/PX4/PX4-Autopilot/blob/5eb13e44480ab4f896c253b77a525f5286ded84a/src/modules/control_allocator/ActuatorEffectiveness/ActuatorEffectivenessTiltrotorVTOL.cpp#L54).
Wouldn't tilting the rotor cause a simultaneous change in force and moment? Why is only moment provided in [ActuatorEffectivenessTilts::addActuators](https://github.com/PX4/PX4-Autopilot/blob/9d7c4b82732413d4bed4891e9fee48e605d732d3/src/modules/control_allocator/ActuatorEffectiveness/ActuatorEffectivenessTilts.cpp#L90)？
@bkueng @sfuhrer

mengchaoheng, 2023-01-19:
And, I have a new Geometry UAV which is fly by the PX4 v1.13.0, I added a mix file to support it. But now the new version of PX4 have change the defualt function of mixing to allocator, can you tell me how can  take off my UAV by the new allocator? what files I should add?

My old .main.mix file is :
```
-------------
M: 2
S: 0 0  10000  10000      0 -10000  10000
S: 0 3       0   20000 -10000 -10000  10000

This mixer is empty.
-------------
M: 2
S: 0 0  -10000  -10000      0 -10000  10000
S: 0 3       0   20000 -10000 -10000  10000

This mixer is empty.
-------------
Z:

This mixer is empty.
-------------
Z:

roll control surfaces
-------------
M: 2
S: 0 1  10000  10000      0 -10000  10000
S: 0 2  -10000  -10000      0 -10000  10000

pitch control surfaces
-------------
M: 2
S: 0 1  10000  10000      0 -10000  10000
S: 0 2  10000  10000      0 -10000  10000
```

and the UAV is like [RC Chinook Bicopter](https://www.youtube.com/watch?v=hfss7nCN40A)

[image]

mengchaoheng, 2023-01-20:
@MaEtUgR  
The torque of RateControl::update is state that is [-1,1] normalized torque vector to apply to the vehicle, but I can't see how to guarantee that [-1,1] range，can you help me ?

sfuhrer, 2023-02-01:
>1.why the getEffectivenessMatrix of ActuatorEffectivenessMCTilt didn't update the aixs of rotor from tilt?

Looks indeed like that could be added. As the tilts on a MC usually don't tilt that much (compared to a VTOL that tilts them 90° down), it doesn't have a big effect though. @bkueng could tell you more, I think he also already testflew a MC tiltrotor with the new allocation. 

>2.What exactly are the physical dimensions (units) of the actuator_motors and actuator_servos output by the allocator?

It's at the moment all unit-less. `u` is the normalized control, so not directly the rotor speed.
BTW, I think the comment is wrong, it should be a linear relation: `Thrust = CT * u`

>And, I have a new Geometry UAV which is fly by the PX4 v1.13.0, I added a mix file to support it. But now the new version of PX4 have change the defualt function of mixing to allocator, can you tell me how can take off my UAV by the new allocator? what files I should add?

Something like that? That's how we set up our bi-copter.
[image]

>By the way, the torque of RateControl::update is state that is [-1,1] normalized torque vector to apply to the vehicle, but I can't see how to guarantee that，can everybody help me ?

Can you be more specific? The torque is normalized between [-1,1], where -1  is meaning something like "maximum available torque in negative direction".