PX4/PX4-Autopilot #20972: cross-compilation on Raspberry Pi fails

christianrauch, 2023-01-19:
## Describe the bug
`make scumaker_pilotpi_default` fails with:
```
[...]/PX4-Autopilot/src/drivers/distance_sensor/lightware_laser_serial/lightware_laser_serial.cpp: In member function ‘virtual void LightwareLaserSerial::Run()’:
[...]/PX4-Autopilot/src/drivers/distance_sensor/lightware_laser_serial/lightware_laser_serial.cpp:311:39: error: ‘ssize_t write(int, const void*, size_t)’ reading 5 bytes from a region of size 4 [-Werror=stringop-overread]
  311 |                         (void)!::write(_fd, &data, strlen(data));
      |                                ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1plus: all warnings being treated as errors

```

## To Reproduce

Follow the instructions at https://docs.px4.io/main/en/flight_controller/raspberry_pi_pilotpi_rpios.html:
1. `git clone https://github.com/PX4/PX4-Autopilot.git --recursive`
2. `cd PX4-Autopilot`
3. `make scumaker_pilotpi_default`

## Expected behavior
Target `scumaker_pilotpi_default` should compile.

dagar, 2023-01-19:
It's just a small warning I can fix tomorrow (or feel free to open a PR). Arguably we should have -Werror disabled by default and only enforced in upstream builds.

christianrauch, 2023-01-19:
I think Werror is ok as long as it is tested on the CI and this kind of code does not get merged. Otherwise, you will not notice when such things sneak into the codebase as these cross-compiling targets seem to be rarely tested.

dagar, 2023-01-21:
This board is already building in CI regularly, the difference is likely the specific compiler/toolchain version.

I'm not getting this error locally, what's your build environment? If you start from a clean state you can share the output of the initial cmake configure.

christianrauch, 2023-01-21:
I am running on Ubuntu 22.04 with the default compilers:
```
-- PX4 version: v1.13.2-10-g9cc2da12de
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.6", minimum required is "3") 
-- PX4 config file: /home/christian/Development/PX4-Autopilot/boards/scumaker/pilotpi/default.px4board
-- PLATFORM posix
-- LINUX y
-- TOOLCHAIN arm-linux-gnueabihf
-- ARCHITECTURE cortex-a53
-- ROMFSROOT px4fmu_common
-- TESTING y
-- PX4 config: scumaker_pilotpi_default
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 11.3.0
-- The C compiler identification is GNU 11.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-linux-gnueabihf-gcc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-linux-gnueabihf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-linux-gnueabihf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- ROMFS: ROMFS/px4fmu_common
-- ROMFS:  Adding boards/scumaker/pilotpi/init/rc.board_defaults -> /etc/init.d/rc.board_defaults
-- Configuring done
-- Generating done
```