PX4/PX4-Autopilot #20981: ROS2/Microdds support with use cases in mavros

Jaeyoung-Lim, 2023-01-22:
## Describe problem solved by the proposed feature
mavros has been the primary interface people have been using to support advanced autonomy use cases. However, since mavros uses mavlink, this has been a major constraint on supporting use cases requiring high performance data links.

Therefore, we want to use the microdds as the primary interface for more advanced autonomous capabilities in the future.

## Describe your preferred solution
This issue has been created to track features that are already supported with `mavros` to be supported in microdds. This way we would encourage people on moving over to the new interface with less regressions.

To list the features that is currently supported with mavros are as the following.
- [ ] Offboard setpoint publishing
- [ ] Mount control / configuration
- [ ] Getting system status information

## Additional context
- Feel free to add features that you care about that were done through mavros, but you would like to see supported by the microdds bridge

russkel, 2023-01-22:
Hi Jaeyoung,

I saw in some of your PRs that you were using `actuator_controls_0` topic to set the actuators in offboard mode (I have been unable to replicate this) yet it's not ticked as complete here. Has it stopped working since #18581 ?

henrykotze, 2023-01-23:
A nice feature in px4 v1.12 with protocol splitter, was that we could send both mavlink-and rtps messages over one serial bus, which enabled us to use already mavlink enabled software. As of latest, we need to convert all the dds messages to mavlink to make use of all other mavlink supported software, delaying our jump to the newest PX4. 

So would be great if microdds could convert the the dds messages to the "normal" mavlink mode messages.

Jaeyoung-Lim, 2023-01-23:
@russkel That was for mavlink based offboard control

actuator_controls are being replaced by thrust_setpoints and torque_setpoints which I havent really tried with microdds

Jaeyoung-Lim, 2023-01-23:
> So would be great if microdds could convert the the dds messages to the "normal" mavlink mode messages

I am not sure if I understand the use case here. Why would you want to translate microdds to mavlink? You can already create mavlink instances to subscribe to certain uORB topics

henrykotze, 2023-01-23:
On our onboard computer (OBC), we use mavlink messages to provide backup coms to our ground station, both for cellular network and satellite. So with protocol splitter we would just forward the mavlink messages to the respective ports and our software on the OBC would than handle the hardware comms to send the mavlink message over the network. It was very nice to use a single serial port for both mavlink and RTPS on both the FMU and OBC.

With microdds, we can still use our ROS2 nodes, but from what I understand, we will need a single node, which subscribes to the  dds messages, convert them to the mavlink messages and send it to the correct port to allow our other software to still be useful. 

Hope this makes more sense.  If not, and your interested we can schedule a call.

dagar, 2023-01-23:
@henrykotze I'm interested in that use case as well, but unfortunately the "protocol splitter" driver had some issues and makes the work flow even more convoluted.

Longer term my goal is to be able to run PX4 modules as actual ROS2 nodes, so if you needed a mavlink instance Linux side you'd simply run one there.

If this truly is a blocker for you and you can't run another serial (or use ethernet or USB) I have another idea for tunneling mavlink output we could probably hack together pretty quickly.

dagar, 2023-01-23:
Short term the "hack" is simply running another mavlink instance where the output goes into a topic that's sent out over microdds, then on the ROS2 side we have a simple node that consumes that topic and spits it out on UDP, etc. 

> Now that I think of it: It should be possible to have both microdds and a mavlink instance sending over ethernet, just over different ports? @dagar

Yes.

henrykotze, 2023-01-23:
Coolbeans! We will probably go the route of sending both other the network. 

Is there any todo list on the  pixhawk 6x? We are looking into using it, but I am hesitant as its quite new? We recently just bought a single one for testing but haven't actually gotten to test it out yet.

Jaeyoung-Lim, 2023-02-21:
- Compatible standard ros2 messages

Jaeyoung-Lim, 2023-04-08:
Currently there doesn't seem to be a good way to pull the mission the vehicle is currently flying in

ozitoune, 2023-04-11:
Hello, i need to control 2 servo left and right via ros2 connected to my v6x by ethernet. The xrce agent and client are well connected...I'm a little confuse do do that there is so many topics !  In witch topic should i publish ?
Actuator_outputs, actuator_servos, vehicule_commande ( do set actuator) ... 
Best regard.
Olivier

laylow13, 2023-04-12:
Hi，I want to directly publish force and torque commands in offboard mode to directly control the drone. What topic should I use? Is this method supported by the firmware?

ozitoune, 2023-04-14:
Hi, 
I think you can publish your command on vehicle_torque_setpoint topic

laylow13, 2023-04-17:
Thank you for your suggestion, I tried but it didn't work well.

ozitoune, 2023-04-17:
perhaps you need to put the px4 in off board control mode before

Noe-Sanchez, 2024-04-02:
@Jaeyoung-Lim any updates on Waypoint push/pull? I'd be happy to contribute if it's still an unsolved issue.