PX4/PX4-Autopilot #20991: V1.13.2 PWM_MAIN_TRIM re-arm update

brunweil, 2023-01-24:
## Describe the bug
save new PWM_MAIN_TRIM d'ont update PWM output in real time but only after disarm re-arm cycle.
## To Reproduce
Steps to reproduce the behavior:
1. Drone switched on '...'
2. QGC V4.2.3/windows connected via Holybro Telemetry Radio
3. Open Vehicle SetUp / Parametres, find PWM_MAIN_TRIM
4. Setup new value e.g. -0.10 for PWM_MAIN_TRIM1 and SAVE the edited param.
5. Nothing happens: for the update of the parameters to occur, it is essential to arm or Disarm/Arm cycle, making the parameter near unusable....
## Expected behavior
as long as I can remember, on the previous version, the update was instantaneous and allowed precise trimming of all actuators...
## Drone (please complete the following information):
- FixedWing
nsh> ver all
HW arch: PX4_FMU_V5
HW type: V500
HW version: 0x00000000
HW revision: 0x00000000
FW git-hash: 46a12a09bf11c8cbafc5ad905996645b4fe1a9df
FW version: Release 1.13.2 (17629951)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 91bece51afbe7da9db12e3695cdbb4f4bba4bc83
Build datetime: Nov 22 2022 09:19:59
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000200000000353338373138511500370029
MCU: STM32F76xxx, rev. Z

## Additional context
Add any other context about the problem here.

dagar, 2023-01-24:
Did you have control allocation enabled (SYS_CTRL_ALLOC=1)? 

It's a px4io, but I actually thought this had been fixed with along with control allocation changes, but have you tried to reproduce on current main? I'm hoping to get a v1.14 stable release out in the not too distant future, so that would be the ideal place to address this if it's not already fixed.

brunweil, 2023-01-24:
No, cause we use an external Mix Table
(implementing Flap on elevon & rudder)
[halfbreath.main.txt](https://github.com/PX4/PX4-Autopilot/files/10492432/halfbreath.main.txt)

brunweil, 2023-01-24:
No trying to reproduce on Main
Trying to solve other matter (dual power monitoring) for hoping near flight ;-)