PX4/PX4-Autopilot #20996: Adding new Airframe file not working

PrometheusTransport, 2023-01-24:
## Describe the bug
Copying existing Caipirinha VTOL airframe config doesn't result in a working vehicle config when loaded on FC and selected via QGC.

New airframe is red in QGC setup, under summary->airframe->airframe type is "setup required", same for summary->airframe->vehicle is "setup required".     also none of the parameter changes in the airframe file are applied

## To Reproduce
cd ROMFS/px4fmu_common/init.d/airframes/

cp 13001_caipirinha_vtol 13015_caipirinha_vtol

change line #3 to be "# @name Test Caipiroshka Duo Tailsitter"

make cubepilot_cubeorange_default

upload firmware to FC

select new airframe in QGC

## Expected behavior
New Vehicle is available in QGC and starts the same as the stock "Caipirinha"

per the docs:

`Adding a configuration is straightforward: create a new config file in the init.d folder (prepend the filename with an unused autostart ID), then build and upload the software.`

## Log Files and Screenshots
not a flight issue

## Additional context
The behavior under 1.11 is as described in the docs.   I tried all of the stable versions of 1.13

It was suggested on discord that editing cmake files would be required, but was not required under 1.11.1.   If the process has changed for adding a new airframe, perhaps the docs are out of date?

tayyabkhalil-313, 2023-01-25:
As per my understanding, adding the name of the airframe in CMake file has always been required. You are required to add the name of airframe file in [this](https://github.com/PX4/PX4-Autopilot/blob/main/ROMFS/px4fmu_common/init.d/airframes/CMakeLists.txt) CMake file before building the code.

PrometheusTransport, 2023-01-25:
I'm certainly willing to believe that is required (and I will test it), it definitely was not required for 1.11.    And if it is required to update the CMakelist file, then this would be a bug against the docs I suppose?   

Perhaps the devs can chime in?

RbtsEvrwhr-Riley, 2023-01-27:
Yep, updating the cmakelist.txt is required.

sfuhrer, 2023-02-14:
@PrometheusTransport if you want it baked into the firmware then yes, you'll need to add it to the cmakelist. Alternatively there is also a workflow with the config file moved to the SD card: https://docs.px4.io/main/en/concept/system_startup.html#customizing-the-system-startup.
I guess you want it to setup multiple vehicle consequently?