PX4/PX4-Autopilot #21014: Errors During Setup of PX4 Dev environment on Ubuntu

ryanpinney, 2023-01-26:
Hello,

I am trying to run the example simulation project with gazebo after setting up the px4 development environment on a dual booted linux computer. I am following this tutorial https://youtu.be/OtValQdAdrU when i run "make px4_sitl jmavsim" the simulation will not start. I am new to linux so any help would be awesome, thanks!

1st error (keeps running after this one because I think it gets through):

Could NOT find gz-transport (missing: gz-transport_DIR)
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:72 (include)
  src/modules/simulation/simulator_mavlink/sitl_targets_gazebo-classic.cmake:2 (find_package)
  src/modules/simulation/simulator_mavlink/CMakeLists.txt:59 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

2nd error (have tried ufw to unblock port):

WARN  [simulator_mavlink] Please ensure port 4560 is not blocked by a firewall.

3rd error:

Exception in thread "main" java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.InternalError: XXX0 profile[2]: GL2 -> profileImpl GL3bc !!! not mapped 
	at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2098)
	at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1976)
	at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1895)
	at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1861)
	at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:80)
	at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:239)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:225)
	at com.jogamp.opengl.GLProfile.getProfileMap(GLProfile.java:2324)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:1042)
	at com.jogamp.opengl.GLProfile.get(GLProfile.java:1071)
	at com.jogamp.opengl.GLProfile.getMaxFixedFunc(GLProfile.java:824)
	at javax.media.j3d.JoglPipeline.initialize(JoglPipeline.java:131)
	at javax.media.j3d.Pipeline.createPipeline(Pipeline.java:92)
	at javax.media.j3d.MasterControl.loadLibraries(MasterControl.java:858)
	at javax.media.j3d.VirtualUniverse.<clinit>(VirtualUniverse.java:267)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:194)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:193)
	at me.drton.jmavsim.Simulator.main(Simulator.java:944)
	... 5 more

nihavend, 2023-01-28:
This bug is related with jogl and fixed by the main developer of the library. PX4 team has to replace the relevant libraries with the fixed ones. It is about the GLProfile java class.

ryanpinney, 2023-01-30:
Using:

cd PX4-Autopilot
make clean
make px4_sitl gazebo

works to run simulation in gazebo

I think the install video is no longer up to date.