PX4/PX4-Autopilot #21017: Stopping Default TCP SITL Listener Prevents Accel 0 and Gyro 0 on UDP

RbtsEvrwhr-Riley, 2023-01-27:
## Describe the bug
On a WSL2 environment, we set up a simulation that required a UDP bridge (mav2xplane UDP https://github.com/RbtsEvrwhr-Riley/mav2xplane). By default it starts a TCP listener for the sim. Stopping this and starting UDP prevents sensors from acquiring.

## To Reproduce

make px4_sitl none
"simulator_mavlink stop"
"simulator mavlink start udp"

Starts the UDP listener.

When it connects, accel 0 and gyro 0 will always provide invalid data, because they will not be updated on the correct ID. Baro and compass work as expected.

## Expected behavior
The TCP sim that never connected should not set up any sensors.

## Log Files and Screenshots
There is no flight, simulated flight could not start.

If you need more screenshots to reproduce I can get them but I need coffee.

## Additional context
The following code block works as a workaround if added to px4-rc.simulator

This is not a fix, it just replaces starting TCP with starting UDP, then everything works.

`# otherwise start simulator (mavlink) module
	simulator_udp_port=$((14560+px4_instance))

	# Check if PX4_SIM_HOSTNAME environment variable is empty
	# If empty check if PX4_SIM_HOST_ADDR environment variable is empty
	# If both are empty use localhost for simulator
	if [ -z "${PX4_SIM_HOSTNAME}" ]; then

		if [ -z "${PX4_SIM_HOST_ADDR}" ]; then
			echo "INFO  [init] PX4_SIM_HOSTNAME: localhost"
			simulator_mavlink start udp $simulator_tcp_port
		else
			echo "INFO  [init] PX4_SIM_HOSTNAME: ${PX4_SIM_HOST_ADDR}"
			simulator_mavlink start udp -u "${PX4_SIM_HOST_ADDR}" "${simulator_tcp_port}"
		fi

	else
		echo "INFO  [init] PX4_SIM_HOSTNAME: ${PX4_SIM_HOSTNAME}"
		simulator_mavlink start udp
	fi`

dagar, 2023-01-27:
If there's any bogus init happening in `simulator_mavlink` before there's a valid connection we can certainly fix it, but I would still suggest the proper fix here is to directly start only the desired interface in the first place.

RbtsEvrwhr-Riley, 2023-01-27:
I cannot find any build option to change the init interface to UDP, having gone through source and had to modify it to allow such. TCP was hardcoded everywhere.

RbtsEvrwhr-Riley, 2023-01-27:
The bug is definitely the init issue that's causing stopping and starting a new interface to break sensors, though; on the other hand, a menuconfig option or something to default UDP would be nice.