PX4/PX4-Autopilot #21033: iris_opt_flow EKF uses GPS instead of flow camera (Gazebo Classic)

MichaelBosello, 2023-01-31:
## Describe the bug
The iris with flow camera is expected to use the flow camera for estimations but still uses the GPS. If GPS is disabled (EKF2_GPS_CTRL 0), the quad will not arm in whatever flight mode.

## To Reproduce
Prepare gazebo env as per instructions
```
git clone https://github.com/PX4/PX4-Autopilot.git --recursive
bash ./PX4-Autopilot/Tools/setup/ubuntu.sh
```

Run gazebo classic with iris opt flow
`make px4_sitl gazebo-classic_iris_opt_flow`

A warn informing that GPS is enabled is shown even though in **1010_gazebo-classic_iris_opt_flow** _EKF2_GPS_CTRL_ is set to 0
`WARN  [ekf2] GPS enabled by EKF2_HGT_REF`

Param show will indeed show _EKF2_GPS_CTRL_ set to 7
```
> param show
EKF2_GPS_CTRL [346,531] : 7
EKF2_HGT_REF [361,546] : 1
```

Change _EKF2_HGT_REF_ and actually set _EKF2_GPS_CTRL_ to 0
```
param set EKF2_HGT_REF 2
param set EKF2_GPS_CTRL 0
```

Now a warning about velocity estimate is shown and arming is prevented
`WARN  [health_and_arming_checks] Preflight Fail: velocity estimate error`

## Expected behavior
It is expected that the quad will arm without GPS as velocity estimation is provided by the flow camera.
Besides, **1010_gazebo-classic_iris_opt_flow** should contain `param set-default EKF2_HGT_REF 2` to conform to the expected behavior

Jaeyoung-Lim, 2023-02-01:
@MichaelBosello I don't think the expectation is correct. The `iris_opt_flow` model is there to test a vehicle with GPS AND Optical flow.

MichaelBosello, 2023-02-01:
@Jaeyoung-Lim for the expectations, why then there is `param set-default EKF2_GPS_CTRL 0 ` inside __1010_gazebo-classic_iris_opt_flow__ ? Who wrote the code indeed tried to remove GPS from the ekf2 fusion without success. 

Apart from the expectations, yes I would like to  test the use case of a vehicle without GPS and only optical flow, but, as stated in the issue, if I disable the GPS the vehicle will not arm anymore (in any of the flight modes).

Jaeyoung-Lim, 2023-02-01:
@MichaelBosello Are you sure that `EKF2_GPS_CTRL 0` disables the gps? The documentation states the opposite: https://docs.px4.io/main/en/advanced_config/parameter_reference.html

[image]

> Apart from the expectations, yes I would like to test the use case of a vehicle without GPS and only optical flow, but, as stated in the issue, if I disable the GPS the vehicle will not arm anymore (in any of the flight modes).

How does it fail? Do you have a log?

MichaelBosello, 2023-02-02:
@Jaeyoung-Lim `EKF2_GPS_CTRL` uses a bitmap so yes 0 means everything is disabled. The documentation says to set the **bit** in position zero (i.e., value 1) to enable lon/lat, not to set 0 to enable (even in QGC if you uncheck everything it puts 0, if you check lon/lat it puts 1, and so on).

The error preventing arming is [`health_and_arming_checks] Preflight Fail: velocity estimate error`. The log file is [here](https://logs.px4.io/plot_app?log=96d7c7d1-aead-4e5a-a884-9a287064a3bc). Thank you for the help.

MichaelBosello, 2023-02-02:
It could be related to #21005

bresch, 2023-02-03:
@MichaelBosello Does it work if you set `param set-default EKF2_HGT_REF 2` before `param set-default EKF2_GPS_CTRL 0` in the config file?

Yes, the preflight failure is probably related to : https://github.com/PX4/PX4-Autopilot/pull/21005

MichaelBosello, 2023-02-03:
@bresch I can set successfully EKF2_HGT_REF 2 and EKF2_GPS_CTRL 0 in the config file, but then the quad will not be able to arm.