PX4/PX4-Autopilot #21035: Adaptation from microRtps to XRCE-DDS

mohamed1915, 2023-02-01:
[image]
Hello, I'm adapting a project that used Micro-RTPS to the new PX4 XRCE-DDS communication protocols. I used @Jaeyoung-Lim offboard example to solve some issues but other issues linger.  The project runs multiple robots each uses PX4 with connection to gazebo and ROS2. I have few questions:

- With micro-RTPS we used a port for each robot, does XRCE-DDS needs the same thing?
- Is there any documentation for the migration to the new PX4, especially the communication part (RTPS->XRCE-DDS)?
-  the node /px4_micro_xrce_dds is not connected to any part of ROS, it only generates the PX4 topics (fmu/in...or fmu/out....). I have publishers and subscribers in python that used to run on rtps, what changes do I need to make?
Thank you,

mrpollo, 2023-02-01:
The documentation is still wip here's the discussion https://github.com/PX4/PX4-user_guide/pull/2094#pullrequestreview-1278584830

Jaeyoung-Lim, 2023-02-01:
> the node /px4_micro_xrce_dds is not connected to any part of ROS, it only generates the PX4 topics (fmu/in...or fmu/out....). I have publishers and subscribers in python that used to run on rtps, what changes do I need to make?

@mohamed1915 What is missing from https://github.com/Jaeyoung-Lim/px4-offboard ?

beniaminopozzan, 2023-02-08:
@mohamed1915 

> With micro-RTPS we used a port for each robot, does XRCE-DDS needs the same thing?

It's not necessary to use different ports (which would require one Agent for each drone). You can use one agent, all clients with the same port but different `session_key_id`

https://github.com/PX4/PX4-Autopilot/blob/fbc80c9bf592d500d5583fdf30278ae7086c6e06/src/modules/microdds_client/module.yaml#L47-L57

ViniciusAbrao, 2023-02-08:
Hi, let me include a question in this issue. Maybe if not related I can create a new one.

In XRCE-DDS client, looks like there is not a mission's service implemented yet. Also, I did not find any service as those implemented by mavros like: arm, set_mode, take off, land ....

Do you know something about it?

thank you

mohamed1915, 2023-02-09:
> @mohamed1915
>
>
> It's not necessary to use different ports (which would require one Agent for each drone). You can use one agent, all clients with the same port but different `session_key_id` [...]

Thank you @beniaminopozzan.

mohamed1915, 2023-02-09:
> Hi, let me include a question in this issue. Maybe if not related I can create a new one.
>
> In XRCE-DDS client, looks like there is not a mission's service implemented yet. Also, I did not find any [...]

This is an interesting question. Because I also can't arm my drones in gazebo from ROS2 nodes. @Jaeyoung-Lim @beniaminopozzan  do you have any ideas?
Thank you,

ViniciusAbrao, 2023-02-12:
> This is an interesting question. Because I also can't arm my drones in gazebo from ROS2 nodes. @Jaeyoung-Lim @beniaminopozzan do you have any ideas? Thank you,

I have uploaded the code to include: arm, take off, offboard flight and land:

https://github.com/ViniciusAbrao/px4_ros2_xrcedds/blob/master/offboard_control.py