PX4/PX4-Autopilot #21043: BARO timeout when running SITL gazebo-classic nolockstep on Rover - main branch

russkel, 2023-02-02:
## Describe the bug
Barometer timeout errors appear in the shell.

```
ERROR [vehicle_air_data] BARO #0 failed:  TIMEOUT!	
INFO  [vehicle_air_data] BARO switch from #0 -> #1
```

## To Reproduce
Steps to reproduce the behavior:
1. Launched SITL from within Docker
2. Observe errors on shell

```
./Tools/docker_run.sh 'make px4_sitl_nolockstep gazebo-classic_rover'
```

## Expected behavior
No barometer errors.

## Log Files and Screenshots
https://logs.px4.io/plot_app?log=0352673a-690e-41a0-93cd-363471be4c7a

## Drone (please complete the following information):
- SITL Gazebo-classic rover

## Additional context
Running up to date main branch as of posting.
Also nuttx commands (ls, ps etc) seem to have stopped working

321thijs123, 2023-05-12:
Have you been able to solve this? I'm running into the same issue while trying to do thrust and torque command with px4_sitl_nolockstep.

russkel, 2023-05-15:
> Have you been able to solve this? I'm running into the same issue while trying to do thrust and torque command with px4_sitl_nolockstep.

Only by turning off the sensors using the `SYS_HAS_BARO` parameter.

matiasguzmanp, 2024-08-20:
Hi! Did you manage to solve this issue? I'm having the exact same problem. The magnetometer also fails.

russkel, 2024-08-22:
Sorry, I stopped using PX4 and started using Ardupilot as it had better support.

closed 2024-08-22 by russkel