PX4/PX4-Autopilot #21055: Preflight Fail: velocity estimate error when enabling flow camera in EKF2

camillo2008, 2023-02-02:
## Describe the bug
In v1.14 when the EKF2_AID_MASK is set to 2, the quad becomes not ready to fly, and a warning about the velocity estimate is shown. This is not the case for v1.13.

## To Reproduce
Steps to reproduce the behavior:
1. Flash current main
2. Chose the airframe and calibrate the sensors
3. Enable the flow sensor and rangefinder with UAVCAN_SUB_FLOW = 1 and SENS_EN_LL40LS = 2
4. Enable the flow sensor with EKF2_AID_MASK = 2

## Expected behavior
The quad should be ready to fly without estimation errors

## Log Files and Screenshots
https://review.px4.io/plot_app?log=0613fcf2-b007-4771-b0b4-5c20456307ef

## Drone:
- Quadcopter with MRO control zero H7
- Hex HereFlow
- Garmin Lite v3

## Additional context
We could exclude a hardware issue as the problem is not present using 1.13. Also, the sensors appear to behave fine in the Mavlink Inspector (OPTICAL_FLOW_RAD, DISTANCE_SENSOR).

May be related to:  #21033

AlexKlimaj, 2023-02-02:
https://github.com/PX4/PX4-Autopilot/issues/20929

https://github.com/PX4/PX4-Autopilot/pull/21005

z1047941150, 2023-02-14:
All my thanks, I meet this problem too.