PX4/PX4-Autopilot #21058: Offboard set_actuator_control_target not working (mavlink)

joaolehodey, 2023-02-02:
## Describe the bug
We are a team of researchers and need direct access to the actuators when working with Rovers/Boats, but this feature does not seem to be working.

We are using Mavsdk to send values between -1 and 1 to the actuators.

I checked and can confirm that the values are being received in the vehicle and being correctly published to the topic with ORB_ID = actuator_controls_0 (as well as actuator_controls_1).

[image]

Therefore values are being published but no response can be seen.

This is being handled by the following function:
[image]

I don't mind correcting the error and creating a PR, but I would like to know what is the best way to fix this issue.
I think that the problem is that 'actuator_controls_0' is not being used and has been replaced by 'vehicle_thrust_setpoint' and 'vehicle_torque_setpoint' as mentioned in issue #20981, but I might be wrong. I have also encountered some other interesting topics namely: 'ORB_ID(actuator_motors)' and 'ORB_ID(actuator_servos)' that might be key to solving this problem.

I would appreciate it if someone could help me understand the issue. And if I am correct in the above statement, what would be the mapping from control groups to 'some other' topic?

## To Reproduce
Steps to reproduce the behavior:
1. Start PX4 SITL Rover Gazebo Simulation (tested both v1.13.2 and v1.14.0-beta1) (lockstep must be disabled -> check https://docs.px4.io/main/en/simulation/ where it shows how to disable it)
2. Open QGC and wait for connection
3. Test the following python script  [off_board_actuator.zip](https://github.com/PX4/PX4-Autopilot/files/10572126/off_board_actuator.zip)
4. Check on console uORB topics actuator_controls_0 as well as actuator_controls_1 (using command: listener actuator_controls_0 in the terminal where px4_sitl is running) where we can see that the correct values are being published.
5. No actuators move and we can see that nothing is being published to the actuators (check qgroundcontrol SERVO_OUTPUT_RAW, ... in the mavlink inspector)

[image]

## Expected behavior
We expect to be able to control the actuators directly.

## Log File
https://logs.px4.io/plot_app?log=533206ce-1228-43f7-8797-027ade652ffa

Thank you for your help.

sfuhrer, 2023-02-08:
@bkueng what would you recommend, how should actuators be controlled directly?

bkueng, 2023-02-14:
I haven't thought it through completely but along the lines:
- for ROS2 / flight modes: use `actuator_motors` and `actuator_servos` directly and add a `vehicle_control_mode` flag to stop `control_allocator` from publishing it if set.
- for MAVLink: use the VEHICLE_CMD_DO_SET_ACTUATOR command (https://github.com/PX4/PX4-Autopilot/blob/main/msg/VehicleCommand.msg#L43)

ozitoune, 2023-04-11:
Sorry, but how to change the flag with ros2, how is the syntax ? ro2 topic pub /fmu/out/vehicle_control_mode ? but what i have to put after this ?
Thanks for your help !

hamishwillee, 2023-09-22:
According to @bkueng This should still work. I hope to test it with MAVSDK next week.

hamishwillee, 2023-10-05:
But it doesn't appear to. @mrpollo - this isn't something that we can fix with "messaging"