PX4/PX4-Autopilot #21069: Multiple uavs sitl with offboard control using ros2

returncode13, 2023-02-06:
I am trying a sitl with multiple uavs and offboard control for each uav  using ros2. I have followed [offboard_example](https://github.com/Jaeyoung-Lim/px4-offboard) and was able to successfuly execute it in the case of a single uav.
But when it comes to multiple uav I can't seem to find any joy.
 
Following are the steps that I have tried.

1. Started micro-ros-agent using 
     `ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888`

2. Started multi sim in gazebo using
    `./Tools/simulations/gazebo-classic/sitl_mutliple_run.sh -n 2`
     This launches a gazebo sim with 2 iris uavs as in the attached image

3. Start QGroundController. 
    I am able to see both uavs there. I can successfully take off either of them and HOLD position.

4. source ros2 and px4_msgs and then list the topics using `ros2 topic list`  which lists the topics as follows

```
/clock
/parameter_events
/performance_metrics
/px4_1/fmu/in/obstacle_distance
/px4_1/fmu/in/offboard_control_mode
/px4_1/fmu/in/onboard_computer_status
/px4_1/fmu/in/sensor_optical_flow
/px4_1/fmu/in/telemetry_status
/px4_1/fmu/in/trajectory_setpoint
/px4_1/fmu/in/vehicle_attitude_setpoint
/px4_1/fmu/in/vehicle_command
/px4_1/fmu/in/vehicle_mocap_odometry
/px4_1/fmu/in/vehicle_rates_setpoint
/px4_1/fmu/in/vehicle_trajectory_bezier
/px4_1/fmu/in/vehicle_trajectory_waypoint
/px4_1/fmu/in/vehicle_visual_odometry
/px4_1/fmu/out/failsafe_flags
/px4_1/fmu/out/sensor_combined
/px4_1/fmu/out/timesync_status
/px4_1/fmu/out/vehicle_attitude
/px4_1/fmu/out/vehicle_control_mode
/px4_1/fmu/out/vehicle_global_position
/px4_1/fmu/out/vehicle_gps_position
/px4_1/fmu/out/vehicle_local_position
/px4_1/fmu/out/vehicle_odometry
/px4_1/fmu/out/vehicle_status
/rosout
```

I understand that for each uav there needs to a separate namespace but as you can see in this case of 2 uavs , there appears to be just the one "px4_1" 

I would like some assistance to move forward 

Images for reference:

micro ros agent launched
[image]

QGCS+ Gazebo with uavs 
[image]

returncode13, 2023-02-06:
I note that the topics fluctuate between `/fmu/....`  and `/px4_1/fmu/..`  . Please see the gif attached showing the ouput of `rqt`

[image]

returncode13, 2023-02-06:
publishing to either topics `fmu` or `px4_1/fmu` results in one of the uavs responding but the flight pattern is awkward. Please see attached image. The blue  line indicates the required flight trajectory but the uav follows the green one.
[image]

Jaeyoung-Lim, 2023-02-06:
@returncode13 I think the problem is that both vehicles are publishing over the same udp port. You might want to check which ports they are using

beniaminopozzan, 2023-02-06:
@returncode13 the issue is in how the instance number is used by the startup script to assign namespaces and session keys to the microdds_client. Currently if the instance number is 0 then no additional namespace is used and the session key is 1. The reason is that this scenario assumes a single vehicle simulation.
If the instance number is greater than 0 then the additional namespace `px4_instance-number` is added and the session key is equal to the instance numeber.

This has been tested in gazebo ignition and garden with manual spawn of the drones.
But never in classic gazebo using `sitl_multiple_runs.sh` (my bad).
And it does not work because instance 0 and 1 will have the same session key, which instead must be unique for all clients.

Waiting for a proper fix, the workaround is pretty simple:

- you trick sitl_multiple_run to start from instance 1 instead of 0. To do so change
  ```sh
  function spawn_model() {
	MODEL=$1
	N=$2 #Instance Number
	X=$3
	Y=$4
	X=${X:=0.0}
	Y=${Y:=$((3*${N}))}
  ```
  into
  ```sh
  function spawn_model() {
	MODEL=$1
	N=$(($2+1) #Instance Number
	X=$3
	Y=$4
	X=${X:=0.0}
	Y=${Y:=$((3*${N}))}
  ```
- you replace https://github.com/PX4/PX4-Autopilot/blob/e153d1defc5efa6fc28679a6336c9c152e9581bf/ROMFS/px4fmu_common/init.d-posix/rcS#L265-L271 such that it always add the additional namespace, something like
  ```sh
  param set XRCE_DDS_KEY $((px4_instance+1))
  microdds_ns="-n px4_$px4_instance"
  ```
  
 @Jaeyoung-Lim , using different udp port is possible but it would require an Agent for each client, which is a kind of waste in simulation.

cosimobromo, 2023-02-09:
@beniaminopozzan I tried to follow your suggestions but I am experiencing some problems in QGroundControl. Spawning 2 vehicles, in QGroundControl they are indicated as "Vehicle 2" and "Vehicle 3" (thus skipping "Vehicle 1"). Plus, when trying to set the Offboard Mode, on the terminal it appears "poll timeout" and I cannot continue on both of them.

beniaminopozzan, 2023-02-09:
@cosimobromo I guess that you used the first method them. It's the expected behavior.
You can use the other one to have the vehicles numbered starting from 1.

The second problem is probably due to the simulation lockstep combined with the fact that you switched to OFFBOARD before starting to publish the setpoints.

junwoo091400, 2023-03-24:
@returncode13 #21183 was merged. Could you try again and check if it resolved the issue?

closed 2025-10-26 by github-actions[bot]