PX4/PX4-Autopilot #21076: Encoding error in px4

arongu321, 2023-02-06:
When I ran catkin build that includes the PX4 package, I get this encoding error for some reason.
[image]
From what I searched up it relates to an issue where the open() function is not compatible with Python 2.x but with Python 3.x though I appreciate if there's another solution to this issue.

arongu321, 2023-02-06:
Ok I managed to get rid of the encoding error though these make errors still persist:
make[2]: *** [events/events_generated.h] Error 1
make[1]: *** [src/lib/events/CMakeFiles/events_header.dir/all] Error 2
make: *** [all] Error 2

junwoo091400, 2023-03-05:
Did you resolve the issue at the end?

arongu321, 2023-03-06:
Yes I did, I basically ran the ubuntu.sh script provided in the Tools/Setup directory of the PX4-Autopilot git repository using:
`bash ./PX4-Autopilot/Tools/setup/ubuntu.sh`
Where . is the catkin workspace src directory.

junwoo091400, 2023-03-25:
Closing as solved, thanks for the answer!

closed 2023-03-25 by junwoo091400