PX4/PX4-Autopilot #21086: FlightTaskAuto landing does not consider landing ground altitude of rally point

KonradRudin, 2023-02-08:
## Description
This is an open issue from the discussion in https://github.com/PX4/PX4-Autopilot/pull/21011.
When performing a RTL to a rally point with a VTOL without a distance sensor, depending on the altitude of the rally point the landing can be a hard landing with a fast speed or a very slow landing. (Potentially affects MC as well). The issue seems to be that although the rally point ground altitude is known in the RTL navigator mode, it is not used in the FlightTaskAuto, where the home altitude is used.
To check, the issue in the code:
As far as i can see, we have the altitude above ground available for a safe point here:
https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/navigator/rtl.cpp#L212
And it gets written into the mission item here:
https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/navigator/rtl.cpp#L546
So the altitude should be available in the position_setpoint_triplet.current for the landing type.
However, in the flightTask, the altitude above ground is evaluated here:
https://github.com/PX4/PX4-Autopilot/blob/main/src/modules/flight_mode_manager/tasks/FlightTask/FlightTask.cpp#L176 and does not take into account the known ground altitudes.

Things to keep in mind:

- For a mission landing we can do exactly the same. (As per https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LAND, the altitude is the ground level)
- If you land at a random spot, the altitude of the position_setpoint_triplet.current is hardcoded to 0 AMSL. This needs to be considered as unknown landing altitude. Also it might be better in this case to set it at least to the lowest point on earth or directly to NaN.

## To Reproduce
Steps to reproduce the behavior:
1. Switch on a drone without a distance sensor.
2. Plan a mission and add some rally points, which are on different altitudes than home.
3. Update RTL behavior to go to closest point including rally points.
4. Fly mission
5. When close to the rally point, trigger a RTL.
6. Watch the speed when landing.

## Expected behavior
If no distance sensor is available, the FlightTaskAuto should use the best guess about the ground altitude of the landing location. In case of the RTL to a rally point other than home, this should be available, where for a landing in an arbitrary position, the ground altitude might not be known.

## Log Files and Screenshots
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/main/en/getting_started/flight_reporting.html)).
- Upload the log to the [PX4 Flight Review](http://logs.px4.io/)
- Share the link to the log (Copy and paste the URL of the log)

Add screenshots to help explain your problem.

## Drone (please complete the following information):
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

## Additional context
Add any other context about the problem here.

kibidev, 2023-02-08:
Great summary of the issue, @KonradRudin.

For the random landing, we might want to add a parameter for selecting behavior. For anyone flying short distance on a flat field, assuming home altitude is the same as landing altitude might be the best behavior.

For other cases, it might depend on the drone/battery whether you want to descend on high speed the entire way to be sure to reach the ground before you run out of battery, while in other cases you want to descend slowly and be sure to get a soft landing.

We could also have an option to use the altitude of the closest rally point (or home or landing waypoint) for a random landing, but unsure if this would be useful enough to justify the added complexity.

kibidev, 2023-02-09:
I have started a fix in #21100. Input is appreciated.