PX4/PX4-Autopilot #21088: Can not use any distance sensor on Holybro Durandall

Cristian-wp, 2023-02-08:
Hello, I am trying to use a distance sensor as altimeter.
I am running 1.13.2 firmware version, and I have followed the instruction reported on [lidar_lite page](https://docs.px4.io/v1.13/en/sensor/lidar_lite.html) for I2C connection without any results. 

EKF2_AID_MASK: 133 (GPS + IMU bias + GPS yaw)
EKF2_RNG_AID: 1 (enabled)
EKF2_TERR_MASK: 1 (range finder)
SENS_EN_LL40LS: I2C
EKF2_MIN_RNG: 0.30
SYS_HAS_NUM_DIST: 1 
EKF2_HGT_MODE: 2 (range sensor)

The strange thing is that even with SYS_HAS_NUM_DIST setted, I do not have any error or warning in QGC regarding the missing sensor(is connected).

As you can see from the image, the distance sensor page does not show up in QGroundControl (v4.2.4).

[image]

After the boot if I run `dmesg` on MAVLink Consolle I get the following output:
```
NuttShell (NSH) NuttX-11.0.0
nsh> dmesg
HW arch: HOLYBRO_DURANDAL_V1
HW type: VD00
HW version: 0x00000000
HW revision: 0x00000000
FW git-hash: 46a12a09bf11c8cbafc5ad905996645b4fe1a9df
FW version: Release 1.13.2 (17629951)
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 91bece51afbe7da9db12e3695cdbb4f4bba4bc83
Build datetime: Nov 22 2022 09:02:55
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000343434363038510b0042002b
MCU: STM32H7[4|5]xxx, rev. Y
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [param] importing from '/fs/mtd_params'
INFO  [parameters] BSON document size 2386 bytes, decoded 2386 bytes (INT32:43, FLOAT:78)
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] data manager file '/fs/microsd/dataman' size is 62560 bytes
rgbled_ncp5623c #0 on I2C bus 1 (external) address 0x39
Loading airframe: /etc/init.d/airframes/6001_hexa_x
INFO  [uavcan] Node ID 1, bitrate 1000000
INFO  [px4io] IO FW CRC match
Board sensors: /etc/init.d/rc.board_sensors
icm20689 #0 on SPI bus 1 rotation 2
bmi088_accel #0 on SPI bus 1 rotation 2
bmi088_gyro #0 on SPI bus 1 rotation 2
WARN  [SPI_I2C] Already running on bus 1
ERROR [SPI_I2C] bmi088: no instance started (no device on bus?)
WARN  [SPI_I2C] Already running on bus 1
ERROR [SPI_I2C] bmi088: no instance started (no device on bus?)
ist8310 #0 on I2C bus 3 (external) address 0xE rotation 10
ms5611 #0 on SPI bus 4
ist8310 #1 on I2C bus 1 (external) address 0xE rotation 10
WARN  [SPI_I2C] ll40ls: no instance started (no device on bus?)
ERROR [heater] Valid SENS_TEMP_ID required
WARN  [SPI_I2C] Already running on bus 1
INFO  [pwm_out] instance: 0, max rate: 100, default: 50, alt: 50
INFO  [init] Mixer: /etc/mixers/hexa_x.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
INFO  [pwm_out] instance: 0, max rate: 100, default: 50, alt: 50
ekf2 [843:237]
Starting Main GPS on /dev/ttyS0
Starting MAVLink on /dev/ttyS1
WARN  [mavlink] offboard mission init failed (-1)
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)

NuttShell (NSH) NuttX-11.0.0
nsh> WARN  [commander] Switching to Altitude is currently not available    
INFO  [logger] [logger] /fs/microsd/log/sess137/log100.ulg    
INFO  [logger] Opened full log file: /fs/microsd/log/sess137/log100.ulg
ERROR [uavcan] couldn't start parameter count: -2
INFO  [gps] u-blox firmware version: SPG 3.01
INFO  [gps] u-blox protocol version: 18.00
INFO  [gps] u-blox module: NEO-M8N-0
WARN  [commander] Switching to Position is currently not available    
INFO  [mavlink] Starting mavlink shell
```
If I check the I2C connection with `i2cdetect` and the sensor does not show up on the second bus as expected.

[image]

If I try to manualy start the sensor with `ll40ls start -X -b 2` command, I get this message from MAVLink console:

[image]

The strange thing is that If I change Durandal, lidar lite, and type of sensor (I had try with a I2C teraranger) is the same extact thing. I can not get any sensor work.
I have already check multiple times cables position(even try to swith them) and the voltage, I have 5V on power (external BEC) and 3.3V on the signals.
I have alredy try to use different QGC and PX4 versions, but nothing changes.

Can you please help me?

mengchaoheng, 2023-05-14:
I have some problem like this:

```Console
NuttShell (NSH) NuttX-11.0.0
$ nsh> dmesg
HW arch: PX4_FMU_V5
HW type: V500
HW version: 0x00000000
HW revision: 0x00000000
FW git-hash: 1c8ab2a0d7db2d14a6f320ebd8766b5ffaea28fa
FW version: Release 1.13.3 (17630207)
FW git-branch: heads/v1.13.3
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 4a1dd8680cd29f51fb0fe66dcfbf6f69bec747cf
Build datetime: May 15 2023 00:18:49
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000200000000203736583756500f00460031
MCU: STM32F76xxx, rev. Z
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [param] importing from '/fs/mtd_params'
INFO  [parameters] BSON document size 1598 bytes, decoded 1598 bytes (INT32:36, FLOAT:44)
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] data manager file '/fs/microsd/dataman' size is 62560 bytes
Loading airframe: /etc/init.d/airframes/4001_quad_x
INFO  [px4io] IO FW CRC match
Board sensors: /etc/init.d/rc.board_sensors
icm20602 #0 on SPI bus 1 rotation 2
icm20689 #0 on SPI bus 1 rotation 2
ERROR [SPI_I2C] bmi055: no instance started (no device on bus?)
ERROR [SPI_I2C] bmi055: no instance started (no device on bus?)
ms5611 #0 on SPI bus 4
ist8310 #0 on I2C bus 3 (external) address 0xE rotation 10
WARN  [SPI_I2C] ist8310: no instance started (no device on bus?)
ERROR [SPI_I2C] pmw3901: no instance started (no device on bus?)
WARN  [SPI_I2C] vl53l1x: no instance started (no device on bus?)
INFO  [pwm_out] instance: 0, max rate: 100, default: 50, alt: 50
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
ekf2 [678:237]
INFO  [pwm_out] instance: 0, max rate: 100, default: 50, alt: 50
Starting Main GPS on /dev/ttyS0
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
Starting MAVLink on /dev/ttyS2
INFO  [mavlink] mode: Onboard, data rate: 5760 B/s on /dev/ttyS2 @ 115200B
INFO  [logger] logger started (mode=all)
INFO  [uavcan] Node ID 1, bitrate 1000000
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] /fs/microsd/log/2023-04-08/10_33_08.ulg    
INFO  [logger] Opened full log file: /fs/microsd/log/2023-04-08/10_33_08.ulg

NuttShell (NSH) NuttX-11.0.0
$ nsh> ERROR [uavcan] couldn't start parameter count: -2
INFO  [commander] Failsafe mode activated    
$ nsh> pmw3901 status
INFO  [SPI_I2C] Not running
$ nsh> vl53l1x status
INFO  [SPI_I2C] Not running
$ nsh> param show SENS_EN_PMW3901
Symbols: x = used, + = saved, * = unsaved
x + SENS_EN_PMW3901 [831,1515] : 1

 918/1979 parameters used.
$ nsh> param show SENS_EN_VL53L1X
Symbols: x = used, + = saved, * = unsaved
x + SENS_EN_VL53L1X [839,1524] : 1

 918/1979 parameters used.
$ nsh> pmw3901 -S start
pmw3901 #0 on SPI bus 5 (external, equal to '-b 1')
$ nsh> vl53l1x start -X
WARN  [SPI_I2C] vl53l1x: no instance started (no device on bus?)
$ nsh> 
```

mengchaoheng, 2023-05-14:
@AlexKlimaj

mengchaoheng, 2023-05-14:
My device on :
```
nsh> i2cdetect -b 4
Scanning I2C bus: 4
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```