PX4/PX4-Autopilot #21089: when ı tried to installation the px4 ı got this error

ErenGurel, 2023-02-08:
-- Configuring incomplete, errors occurred!
See also "/home/eren/src/Firmware/build/px4_sitl_default/build_gazebo-classic/CMakeFiles/CMakeOutput.log".
See also "/home/eren/src/Firmware/build/px4_sitl_default/build_gazebo-classic/CMakeFiles/CMakeError.log".
[512/711] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_main.cpp.o
FAILED: external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-configure 
cd /home/eren/src/Firmware/build/px4_sitl_default/build_gazebo-classic && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic && /usr/bin/cmake -E touch /home/eren/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-configure
[524/711] Performing build step for 'libmicroxrceddsclient_project'
-- libmicroxrceddsclient_project build command succeeded.  See also /home/eren/src/Firmware/build/px4_sitl_default/src/modules/microdds_client/src/libmicroxrceddsclient_project-stamp/libmicroxrceddsclient_project-build-*.log
[530/711] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_main.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:227: px4_sitl_default] Error 1

ErenGurel, 2023-02-08:
previous

~/src/Firmware$ DONT_RUN=1 make px4_sitl_default gazebo
[431/711] Performing configure step for 'sitl_gazebo-classic'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem 
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Looking for ignition-math6 -- found version 6.13.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-transport8 -- found version 8.3.0
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Checking for module 'uuid'
--   Found uuid, version 2.34.0
-- Looking for ignition-msgs5 -- found version 5.10.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.13.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.2.0
-- Looking for ignition-msgs5 -- found version 5.10.0
-- Looking for ignition-common3 -- found version 3.14.2
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.14.2
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-fuel_tools4 -- found version 4.6.0
-- Searching for dependencies of ignition-fuel_tools4
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.2
-- Checking for module 'libzip'
--   Found libzip, version 1.5.1
-- Looking for ignition-common3 -- found version 3.14.2
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-msgs5 -- found version 5.10.0
-- Checking for module 'gstreamer-1.0 >= 1.0'
--   No package 'gstreamer-1.0' found
-- Checking for module 'gstreamer-base-1.0 >= 1.0'
--   No package 'gstreamer-base-1.0' found
-- Checking for module 'gstreamer-app-1.0 >= 1.0'
--   No package 'gstreamer-app-1.0' found
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
[... 68 lines cut ...]
    linked by target "gazebo_user_camera_plugin" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_gst_camera_plugin" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "mav_msgs" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_controller_interface" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_imu_plugin" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_mavlink_interface" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_motor_model" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_camera_manager_plugin" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_magnetometer_plugin" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_barometer_plugin" in directory /home/eren/src/Firmware/Tools/simulation/gazebo-classic/sitl_gazebo-classic
[... cut ...] ssic-configure
[524/711] Performing build step for 'libmicroxrceddsclient_project'
-- libmicroxrceddsclient_project build command succeeded.  See also /home/eren/src/Firmware/build/px4_sitl_default/src/modules/microdds_client/src/libmicroxrceddsclient_project-stamp/libmicroxrceddsclient_project-build-*.log
[530/711] Building CXX object src/modules/mavlink/CMakeFiles/modules__mavlink.dir/mavlink_main.cpp.o
ninja: build stopped: subcommand failed.
make: *** [Makefile:227: px4_sitl_default] Error 1

ViniciusAbrao, 2023-02-13:
make px4_sitl gazebo_iris
(Note that make px4_sitl gazebo no longer works).

JaimeParker, 2023-03-01:
> make px4_sitl gazebo_iris (Note that make px4_sitl gazebo no longer works).

info is
```
WARNING gazebo_iris target DEPRECATED, please use gazebo-classic_iris
```
and I sitll got this error

```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_APP_LIBRARIES (ADVANCED)
    linked by target "AdvancedLiftDragPlugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "ForceVisual" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_airship_dynamics_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_pose_sniffer_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "sensor_msgs" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_airspeed_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_groundtruth_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "physics_msgs" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "std_msgs" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_usv_dynamics_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_video_stream_widget" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_gps_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_opticalflow_mockup_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_irlock_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_parachute_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_lidar_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_multirotor_base_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_wind_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_opticalflow_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_uuv_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_drop_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "nav_msgs" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_sonar_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_vision_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_user_camera_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_gst_camera_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "mav_msgs" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_controller_interface" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_imu_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_mavlink_interface" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_motor_model" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
    linked by target "gazebo_camera_manager_plugin" in directory /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic
[... cut ...]
cd /home/hazyparker/Applications/PX4-Autopilot/build/px4_sitl_default/build_gazebo-classic && /usr/local/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /home/hazyparker/Applications/PX4-Autopilot/Tools/simulation/gazebo-classic/sitl_gazebo-classic && /usr/local/bin/cmake -E touch /home/hazyparker/Applications/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo-classic/sitl_gazebo-classic-configure
ninja: build stopped: subcommand failed.
make: *** [Makefile:232: px4_sitl] Error 1

```

I'm running on ubuntu20.04,it's said that 18,04 is better.

FrancSerra, 2023-03-09:
Hey @JaimeParker, I'm facing the exact same problem. How did you fix it? Could you please give me a hand?
Thanks in advance

Jaeyoung-Lim, 2023-03-09:
@FrancSerra @JaimeParker @ErenGurel 
The log shows exactly what package you are missing.
```
-- Checking for module 'gstreamer-1.0 >= 1.0'
-- No package 'gstreamer-1.0' found
-- Checking for module 'gstreamer-base-1.0 >= 1.0'
-- No package 'gstreamer-base-1.0' found
-- Checking for module 'gstreamer-app-1.0 >= 1.0'
-- No package 'gstreamer-app-1.0' found
```
You can install the package using `apt`