PX4/PX4-Autopilot #21090: Gazebo dependency updates

mrpollo, 2023-02-08:
We need to update the `ubuntu.sh` install script for Gazebo dependencies

Ubuntu 22.04 needs o install Gazebo Garden, and never install Gazebo Classic

**NOTE:** We should still support Gazebo Classic in a docker environment

For reference https://github.com/PX4/PX4-user_guide/issues/2265

hamishwillee, 2023-03-01:
> NOTE: We should still support Gazebo Classic in a docker environment

You mean "Gazebo Classic" on Ubuntu 22.04 in a docker environment? Why?

The script needs to run in both local and docker environments and produce the same toolchain - that is why we created it :-).
If you need Gazebo Classic to be something that can be installed on Ubuntu 22.04 then that should be something that happens based on a flag. I wouldn't bother unless there are tests that depend on it.

hamishwillee, 2023-03-03:
Fixed

closed 2023-03-03 by hamishwillee