PX4/PX4-Autopilot #21092: RC Failsafe triggering in Mission mode with COM_RCL_EXCEPT to 0

ryanjAA, 2023-02-09:
On recent Main

COM_RCL_EXCEPT set to 0 and it was in mission mode when lost RC - failsafe was still triggered. 

https://review.px4.io/plot_app?log=7939dcb5-2a7d-4a49-ac00-d5588ec22934

sfuhrer, 2023-02-13:
COM_RCL_EXCEPT set to 0 will not except the RC loss in mission, for that the bit 0 would need to be set (so the parameter be at value 1). Could there have been some miss-interpretation of the parameter description @ryanjAA ?

ryanjAA, 2023-02-13:
I’m not sure if I’m not understanding. 

The param definition is: Specify modes in which RC loss is ignored and the failsafe action not triggered.

0=Mission so shouldn’t it ignore RC failsafe in mission?

ryanjAA, 2023-02-13:
Ok strange. I pulled the definition from the param page. So how do we ensure they match up? 
[image]

ryanjAA, 2023-02-13:
Also, so what is 0 then? Don’t ever ignore?

sfuhrer, 2023-02-13:
0 is don't ever ignore, 1 is ignore in Mission, 3 is ignore in Mission and Hold, 7 is ignore in Mission, Hold and Offboard etc. Which parts of the param description don't match up?

ryanjAA, 2023-02-13:
Look at the bitmask in the image I uploaded above. It’s one number off.

MaEtUgR, 2023-02-13:
@ryanjAA I visualized some example cases from the documentation screenshot in https://github.com/PX4/PX4-Autopilot/issues/21092#issuecomment-1428065723 which can be found here: https://docs.px4.io/main/en/advanced_config/parameter_reference.html#COM_RCL_EXCEPT
|Bit 0|Bit 1|Bit2|Parameter Value|Result|
|---|---|---|---|---|
|1|0|0|1|No RC loss in mission mode|
|0|1|0|2|No RC loss in hold mode|
|0|0|1|4|No RC loss in offboard mode|
|1|1|0|3|No RC loss in mission and hold mode|
|1|1|1|7|No RC loss in mission, hold and offboard mode|

So in the shared log with `COM_RCL_EXCEPT, 0` there will be an RC loss reaction in mission mode which also happens as expected. If it should be skipped the value of the parameter would need to be 1,3 or 7 depending on whether it should also be skipped in other modes.

ryanjAA, 2023-02-13:
Thanks @MaEtUgR and @sfuhrer. That does make sense but then we need to update the docs to make them more clear, even if they are technically speaking already correct. 

Your breakout is not only helpful and while definitely over-kill for in QGC (since the box selection it totally sufficient), it is a good example for some simplified variation in the docs (even if just the output). This would be helpful as by just a quick look at the definitions makes you come to the conclusion I did. Aka 0=Mission.

Much appreciated - closing but @hamishwillee what do you think? I'm trying to avoid someone looking at that and setting it incorrectly without looking at what Matthias wrote.

closed 2023-02-13 by ryanjAA