PX4/PX4-Autopilot #21095: microRTPS to XRCE in Pixhawk 5X

gustavotbarros, 2023-02-09:
Hi folks,

Recently I installed the newest version of Autopilot PX4 and I would like to know how do I build the code with the possibility to use RTPS?

In another version of PX4 Autopilot I could do this with make px4_fmu-v5x_rtps, but now this new release doesn't have it anymore. Any suggestions? I have built the default firmware and can't find the RPTS_CONFIG parameter.

Has anyone tested the XRCE communication on a real drone?

beniaminopozzan, 2023-02-13:
Hi @gustavotbarros , if you want to use microRTPS you need to roll back to version `1.13`. If you want to try microdds then please refer to #21120 :)
I personally used it only in simulation, but any feedback on tests on real drones would be greatly appreciated!

gustavotbarros, 2023-02-15:
Hi, alright

Ive read that some commands like arming the drone, changing flight mode and so on were not implemented in the latest version yet. when this is implemented, i will give it a try.

beniaminopozzan, 2023-02-15:
Even microRTPS did not have "some commands like arming the drone, changing flight mode and so on" implemented, you had to send the appropriate messages ros2 messages (`vehicle_command`, ecc).
microXRCE DDS changes the low level protocol but from the user POV everything is still the same, just different commands to start the bridge.

gustavotbarros, 2023-02-15:
Ah ok, I'll try to do some tests with micro-xrce-dds then. I have one question, is it possible to use microRTPS with Pixhawk 6X? By changing the file .px4board? I have already done that with a CubeOrange.

beniaminopozzan, 2023-02-17:
Yes, you can. I never did it personally but you need to

- As you said, enable microRTPS and disable microDDS (they can't coexist) in .px4board. Use a tag or a release branch that is compatible with both pixhawk6x (not too old) and microRTPS (not too recent), like `v1.13.2` or `release/1.13`  (https://docs.holybro.com/autopilot/pixhawk-6x/supported-firmware). The entries you need to add / modify are
  ```
  CONFIG_MODULES_MICRODDS_CLIENT=n
  CONFIG_MODULES_MICRORTPS_BRIDGE=y
  ```
  
- flash the firmware and configure the microRTPS bridge.

junwoo091400, 2023-03-27:
Was it resolved? @gustavotbarros

gustavotbarros, 2023-03-29:
@junwoo091400 Hi, not yet. I'm updating my codes to the new px4_msgs and topics.

But in SITL I've already noticed that I am not able to send the yaw through TrajectorySetpoint messages.

Has anyone ever flown in Offboard Mode with v1.14?

beniaminopozzan, 2023-03-30:
@gustavotbarros 

> Has anyone ever flown in Offboard Mode with v1.14?

Last time I flown in offboard with v1.14-beta1
Holybro Pixhawk 4 mini connected via serial to a Raspberry Pi 4

beniaminopozzan, 2023-03-31:
@gustavotbarros

> Is it possible to use MicroXRCE with v1.13.2 or v1.13.3?

No, it is not possible. See first note in the [ROS 2 - PX4 documentation](https://docs.px4.io/main/en/ros/ros2_comm.html#ros-2-user-guide)