PX4/PX4-Autopilot #21117: Installing PX4-Autopilot for M1 Mac Users Running Parallels Ubuntu 20.04 Virtual Machine

OliverHeilmann, 2023-02-12:
# Description
As per the title, I have been trying to install PX4-Autopilot so that I can use the Gazebo Rover SITL environment with MavSDK. I am using a Macbook Pro with Apple Silicon (M1 chip) and have a Parallels VM instance running Ubuntu 20.04. 

```sh
Linux ubuntu 5.15.0-53-generic #59~20.04.1-Ubuntu SMP Thu Oct 20 15:10:27 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```

I have managed to get the setup working, following issues including #19210  and #21014, but it was not as straight forward I was hoping. 

I am opening up a **Feature Request** as a means to capture the steps I went through and link the associated resources I used, as well as to ask the PX4 developer community if we can have some clearer documentation for others in my situation. Hopefully, this post will help other developers experiencing the same issue in the meantime.

# Installation
_Note that this post assumes that readers already have a working Parallels (or similar) VM installed._

Clone the repository as normal into the desired location.
```sh
git clone https://github.com/PX4/PX4-Autopilot.git --recursive
```

Next, you need to delete [two lines](https://github.com/PX4/PX4-Autopilot/blob/edb6c635d5cd3d10929fd77e8bab37a4c9f383e0/Tools/setup/ubuntu.sh#L125) in the ```ubuntu.sh``` file located in the ```setup``` folder.

```sh
# delete the two lines (lines 125, 126) in ubuntu.sh
# 		g++-multilib \
#		gcc-multilib \
nano PX4-Autopilot/Tools/setup/ubuntu.sh
```

Install the ARM-Linux gcc and g++ multilib versions.

```sh
sudo apt-get install gcc-multilib-arm-linux-gnueabi gcc-multilib-arm-linux-gnueabihf g++-multilib-arm-linux-gnueabi g++-multilib-arm-linux-gnueabihf
```

The rest of the instructions shown below are as normal and described in the standard instructions [here](https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html).
```sh
bash ./PX4-Autopilot/Tools/setup/ubuntu.sh
sudo reboot

# in PX4-Autpilot directory
make px4_sitl gazebo_rover
```

# Resources Used
1. I found [mengchaoheng](https://github.com/mengchaoheng)'s post to be useful in explaining how developers should 'delete g++-multilib and gcc-multilib lines'.
2. I ended up following the steps from this [YouTube video](https://www.youtube.com/watch?v=YHX5QqF6-Nw).
3. Official [PX4-Autopilot YouTube video](https://www.youtube.com/watch?v=OtValQdAdrU) describing the installation instructions.
4.  The PX4-Autopilot documentation instructions [here](https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html)

Flyingtux, 2023-03-25:
Hello, Thanks for the tuto. I followed it on my M1 64GB with parallels pro 18 on a fresh ubuntu 18.04 32GB but I had some trouble.

which version did you checkout ?

on current version (25/03/2023), gazebo targets seems not work anymore and jmavsim version crashed
on previous versions like 1.13.2 I have this issue : 

jpdumont@macuntu2004:~/PX4-Autopilot$make px4_sitl gazebo_rover
ninja: error: '../../src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py', needed by 'mavlink/development/development.h', missing and no known rule to make it
make: *** [Makefile:235: px4_sitl] Error 1

Thanks in advance

Flyingtux, 2023-03-25:
Hello,

Ok I solved my problem, I was using the v1.11.3.
My problem was a bad init of git when I did :
```
git checkout v1.11.3
git submodule init
git submodule update --recursive
```
the good ways is : 
```
git checkout v1.11.3
git submodule update --init --recursive
```
Regards,

OliverHeilmann, 2023-03-26:
Hi @Flyingtux, thanks for posting your solution! Answering your initial question anyway, I am using ```v1.11.3``` as per the screenshot shown below. 

[image]

My experience of setting up the PX4 SITL environment didn't require me to checkout any specific version (I think the version I have was the default). One thing I did note was that you are using Ubuntu 18.04 whereas I was using 20.04. PX4 documentation says both should be fine but just wanted to highlight this difference.

Another noteworthy thing is that you might need to turn of 3D/ Graphics/ GPU Acceleration while using Gazebo. Some users experience a 'black screen' if not. Generally this slows down the SITL environment but, as you are using an M1 Pro, I am sure it will be fine regardless. See setting option below (click on ```Advanced``` and ensure that this is disabled).

[image]

Can you confirm that the Gazebo and JMavSim simulators are working as you expect now?

I will keep this issue open to allow others to post if they had any other issues following these steps.

Flyingtux, 2023-03-27:
Hello Oliver,

Nice to meet you,
Thanks for your answers/advises

Actually I tested the 3 last versions of ubuntu LTS (18, 20 and 22) to see
which was better.
The only one really working from v1.11.3 to the last current version of PX4
is the 20.04,

My goal was to find the best/easiest solution to develop on my MacM1 (macos
native/parallels Ubuntu/macos dockers...)
For now, I'm thinking to invest in VOXL2 platforms from Modalai based on
Qualcomm RB5. But their docker tool-chains are only build available for
amd64 platform.
Fortunately, they seem to work with Docker under Macos/Rosetta, but not
under parallels Ubuntu. so I will have to jump from Parallels-Ubuntu and
MacOs-Docker-Rosetta but it should be simple with the shared folder

Not easy to work with the Apple ARM architecture. The superb silence of the
M1 has its price ;-).

I did test a lot Gazebo yet, I just noticed that it didn't like big screen
like 3440x1440

Regards,

Jean-Pierre

Flyingtux, 2023-03-27:
Re Hello,
I didn't test jmavsim, I don't like java at all. I already found this
techno slow and sluggish, so not good for real-time applications like sitl
simulations.
And we have already a virtual machine with parallels that slows us down !

Regards,

Jean-Pierre

OliverHeilmann, 2023-05-01:
Hi @Flyingtux, if you're looking into simulators and concerned about performance, I'd have a look at WeBots. I've used it with good success and it uses much less compute power than other simulators such as Gazebo. I think the UI is quite intuitive, as is the nodal structure of all the in-world assets (docs are good too). 

Look at this [comparison between Gazebo, CoppeliaSim, MORSE and WeBots](https://www.sciencedirect.com/science/article/abs/pii/S1569190X22001046) paper for some further insight.

Flyingtux, 2023-05-02:
Hi Oliver,

Thanks, interesting for robotics application ;-)

LongLiveForFreedom, 2024-04-11:
> # Description
> As per the title, I have been trying to install PX4-Autopilot so that I can use the Gazebo Rover SITL environment with MavSDK. I am using a Macbook Pro with Apple Silicon (M1 chip) and have a Parallels VM instance running Ubuntu 20.04.
> 
> ```shell
> Linux ubuntu 5.15.0-53-generic #59~20.04.1-Ubuntu SMP Thu Oct 20 15:10:27 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
> ```
> 
> I have managed to get the setup working, following issues including #19210 and #21014, but it was not as straight forward I was hoping.
> 
> I am opening up a **Feature Request** as a means to capture the steps I went through and link the associated resources I used, as well as to ask the PX4 developer community if we can have some clearer documentation for others in my situation. Hopefully, this post will help other developers experiencing the same issue in the meantime.
> 
> # Installation
> _Note that this post assumes that readers already have a working Parallels (or similar) VM installed._
> 
> Clone the repository as normal into the desired location.
> 
> ```shell
> git clone https://github.com/PX4/PX4-Autopilot.git --recursive
> ```
> 
> Next, you need to delete [two lines](https://github.com/PX4/PX4-Autopilot/blob/edb6c635d5cd3d10929fd77e8bab37a4c9f383e0/Tools/setup/ubuntu.sh#L125) in the `ubuntu.sh` file located in the `setup` folder.
> 
> ```shell
> # delete the two lines (lines 125, 126) in ubuntu.sh
> # 		g++-multilib \
> #		gcc-multilib \
> nano PX4-Autopilot/Tools/setup/ubuntu.sh
> ```
> 
> Install the ARM-Linux gcc and g++ multilib versions.
> 
> ```shell
> sudo apt-get install gcc-multilib-arm-linux-gnueabi gcc-multilib-arm-linux-gnueabihf g++-multilib-arm-linux-gnueabi g++-multilib-arm-linux-gnueabihf
> ```
> 
> The rest of the instructions shown below are as normal and described in the standard instructions [here](https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html).
> 
> ```shell
> bash ./PX4-Autopilot/Tools/setup/ubuntu.sh
> sudo reboot
> 
> # in PX4-Autpilot directory
> make px4_sitl gazebo_rover
> ```
> 
> # Resources Used
> 1. I found [mengchaoheng](https://github.com/mengchaoheng)'s post to be useful in explaining how developers should 'delete g++-multilib and gcc-multilib lines'.
> 2. I ended up following the steps from this [YouTube video](https://www.youtube.com/watch?v=YHX5QqF6-Nw).
> 3. Official [PX4-Autopilot YouTube video](https://www.youtube.com/watch?v=OtValQdAdrU) describing the installation instructions.
> 4. The PX4-Autopilot documentation instructions [here](https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html)

thank you for your sharing! It worked for me!