PX4/PX4-Autopilot #21120: New PX4/ROS Interface Support Thread

mrpollo, 2023-02-13:
Hey folks please drop any issues or questions on this thread and we will try to address them, just be aware that **this feature is still in beta until it's part of a stable release.**

ViniciusAbrao, 2023-02-13:
Working with real hardware (pixhawk 6C), looks like the client is not automatically started... Does anybody know how start the microdds_client ? is it via mavlink? I am trying to do it but if I work with mavlink I can not simultaneously work with the xrce-dds birdge !!

I tried to open the mavlink console in the QGroundControl, start the client, so I close the QGround and in the sequence I start the Agent. Following it I just have the following in the agent terminal:

[1676258724.774920] info | TermiosAgentLinux.cpp | init | running... | fd: 3
[1676258724.775022] info | Root.cpp | set_verbose_level | logger setup | verbose_level: 6

Looks like there is not any ROS topic being published. How do you manage it to see the topics?

ViniciusAbrao, 2023-02-13:
@bperseghetti thank you ... I was trying to connect via USB:

In the MavlinkConsole: microdds_client start -t serial -d /dev/ttyACM0 -b 921600

In my laptop: ./MicroXRCEAgent serial --dev /dev/ttyACM0 -b 921600
also I tested with micro-ros agent.

So, the results are as I mentioned before, the client started, the agent started, but I can not see any Topic. I am going to test it with serial /dev/ttyS3 in raspberry pi and see the results using the TELEM1.

ViniciusAbrao, 2023-02-14:
@beniaminopozzan I have tested it in TELEM2, with raspberrypi, and it worked as expected. Thank you for your attention. 

XRCE_DDS_0_CFG = TELEM2
In the MavlinkConsole: microdds_client start -t serial -d /dev/ttyS3 -b 921600
In raspberrypi: sudo MicroXRCEAgent serial --dev /dev/serial0 -b 921600

beniaminopozzan, 2023-02-14:
@ViniciusAbrao Did it not started autonomously once you set `XRCE_DDS_0_CFG = TELEM2`? Was `microdds_client start -t serial -d /dev/ttyS3 -b 921600` necessary?

ViniciusAbrao, 2023-02-14:
@beniaminopozzan I double checked it now as following:

1- I turned off the raspberry and the pixhawk and in the sequence turned on both again.
2- In the QGroundControl - MavlinkConsole: microdds_client status

I can see that it is already running, it means, it is not necessary to start it again. So maybe the reason is because I set XRCE_DDS_0_CFG = TELEM2 or because I already started it in the past (I do not know if it is related to automatically change some other parameter), the fact is that is not necessary to start it again.

beniaminopozzan, 2023-02-14:
@ViniciusAbrao that is the expected behavior, `XRCE_DDS_0_CFG = TELEM2` is expected to automatically start the client with the baudrate given by `SER_TEL2_BAUD`. As long as you don't need to pass additional arguments to microdds_clients (i.e. a custom namespace) you can do everything using just the parameters.

ViniciusAbrao, 2023-02-14:
@beniaminopozzan great!!! ... so maybe it would be interesting to mention that in the documentation section "XRCE-DDS Client Setup for Flight Controllers".

ozitoune, 2023-02-15:
Sory for my question but i'm newbee, how to pass this paramaters ? via parameters in qgroundcontrol ?

Is it impossible to do this work via usb and ttyACM0 ? 

@beniaminopozzan if i understant if i put  this parameter `XRCE_DDS_0_CFG = TELEM2` i don't need to start the client manually and it start automaticly whit the baudrate indicate in `SER_TEL2_BAUD` ? Is this correct ?

ViniciusAbrao, 2023-02-15:
> Sory for my question but i'm newbee, how to pass this paramaters ? via parameters in qgroundcontrol ?

yes, you can do it in Qgroundcontrol

> Is it impossible to do this work via usb and ttyACM0 ?

WIth Qground the rapsberry will be connecter with laptop via USB, but as I explained you in the link belllow you need a companion computer:

https://github.com/PX4/PX4-Autopilot/issues/20634#issuecomment-1431443324

> @beniaminopozzan if i understant if i put this parameter `XRCE_DDS_0_CFG = TELEM2` i don't need to start the client manually and it start automaticly whit the baudrate indicate in `SER_TEL2_BAUD` ? Is this correct ?

yes, as we were talking in the comments here above.

ViniciusAbrao, 2023-02-17:
@beniaminopozzan I can see here:

https://github.com/PX4/PX4-user_guide/blob/d88460a7594d3930f621d645b461750a29d61cad/en/simulation/gazebo_vehicles.md#quadrotor

That would have an option to make with depth camera:

make px4_sitl gazebo_iris_depth_camera

But, here in the airframes, there is not the correspondent build target:

https://github.com/PX4/PX4-Autopilot/tree/main/ROMFS/px4fmu_common/init.d-posix/airframes

Also I can see there is the model, but without the gazebo's plugin implemented in the sdf for the RGBD camera:

https://github.com/PX4/PX4-SITL_gazebo-classic/blob/9343aaf4e275db48fce02dd25c5bd8273c2d583a/models/iris_depth_camera/iris_depth_camera.sdf

Is the gazebo simulation with depth camera in development or should I use a different make?

beniaminopozzan, 2023-02-17:
@ViniciusAbrao could you open a new issue for this? Let's keep this thread for the micro XRCE-DDS interface :)

ViniciusAbrao, 2023-02-17:
> @ViniciusAbrao could you open a new issue for this? Let's keep this thread for the micro XRCE-DDS interface :)

Ok @beniaminopozzan , I did it here:

https://github.com/PX4/PX4-user_guide/issues/2283

mrpollo, 2023-03-28:
Hey everyone I'm closing this issue, our docs have been merged thanks everyone for your help https://docs.px4.io/main/en/ros/ros2.html

if you still have issues with those docs please open a new issue

closed 2023-03-28 by mrpollo