PX4/PX4-Autopilot #21125: Gazebo simulation compiling error after upgraded OSX from BigSur to the latest Ventura

kevinuav, 2023-02-14:
## Describe the bug
. I am using mac to develop px4's firmware(v1.14.0-beat). It did well on my old system BigSur . But not after I upgraded it to the latest OSX Ventura v13.2. 
## To Reproduce
When compiling for gazebo, it reports the issues like this:
```
/Users/kevin/PX4-Autopilot/src/lib/sensorcalibration/Utilities.hpp:91:2: fatal error: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
        sprintf(str, "CAL%s%u_%s", sensor_type, instance, cal_type);
```
        
 It seems like that the apple's command line doesn't support the function sprintf() anymore. Although the very same codes can be compiled successfully.

## Log Files and Screenshots
Here is the full output:
```
% make px4_sitl gazebo_plane
-- PX4 version: v1.13.0-1-g49c4ca0620
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.16", minimum required is "3") 
-- PX4 config file: /Users/kevin/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is AppleClang 14.0.0.14000029
-- The C compiler identification is AppleClang 14.0.0.14000029
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- ROMFS: ROMFS/px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/kevin/PX4-Autopilot/build/px4_sitl_default
[0/868] git submodule src/drivers/gps/devices
[4/868] git submodule src/modules/mavlink/mavlink
[11/868] git submodule Tools/sitl_gazebo
[12/868] Generating Mavlink developmen...ssage_definitions/v1.0/development.xml
Validating /Users/kevin/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Parsing /Users/kevin/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /Users/kevin/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Parsing /Users/kevin/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Validating /Users/kevin/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Parsing /Users/kevin/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Validating /Users/kevin/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Parsing /Users/kevin/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Merged enum MAV_CMD
Merged enum MAV_CMD
Found 230 MAVLink message types in 4 XML files
Generating C implementation in directory /Users/kevin/PX4-Autopilot/build/px4_sitl_default/mavlink/development
Generating C implementation in directory /Users/kevin/PX4-Autopilot/build/px4_sitl_default/mavlink/standard
Generating C implementation in directory /Users/kevin/PX4-Autopilot/build/px4_sitl_default/mavlink/common
Generating C implementation in directory /Users/kevin/PX4-Autopilot/build/px4_sitl_default/mavlink/minimal
Copying fixed headers for protocol 2.0 to /Users/kevin/PX4-Autopilot/build/px4_sitl_default/mavlink
[22/868] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is AppleClang 14.0.0.14000029
[... 191 lines cut ...]
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
1 error generated.
[305/868] Building CXX object src/lib/sensor_calibration/CMakeFiles/sensor_calibration.dir/Magnetometer.cpp.o
FAILED: src/lib/sensor_calibration/CMakeFiles/sensor_calibration.dir/Magnetometer.cpp.o 
/usr/local/bin/ccache /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"sensor_calibration\" -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I/Users/kevin/PX4-Autopilot/boards/px4/sitl/src -I/Users/kevin/PX4-Autopilot/platforms/posix/src/px4/common/include -I/Users/kevin/PX4-Autopilot/build/px4_sitl_default -I/Users/kevin/PX4-Autopilot/build/px4_sitl_default/src/lib -I/Users/kevin/PX4-Autopilot/platforms/posix/src/px4/generic/generic/include -I/Users/kevin/PX4-Autopilot/platforms/common -I/Users/kevin/PX4-Autopilot/platforms/common/include -I/Users/kevin/PX4-Autopilot/src -I/Users/kevin/PX4-Autopilot/src/include -I/Users/kevin/PX4-Autopilot/src/lib -I/Users/kevin/PX4-Autopilot/src/lib/matrix -I/Users/kevin/PX4-Autopilot/src/modules -I/Users/kevin/PX4-Autopilot/platforms/posix/include -I/Users/kevin/PX4-Autopilot/build/px4_sitl_default/external/Install/include -O2 -g -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fcolor-diagnostics -fdiagnostics-absolute-paths -Qunused-arguments -Wno-c99-designator -Wno-unknown-warning-option -Wno-unused-const-variable -Wno-varargs -fno-exceptions -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -fno-rtti -std=gnu++14 -MD -MT src/lib/sensor_calibration/CMakeFiles/sensor_calibration.dir/Magnetometer.cpp.o -MF src/lib/sensor_calibration/CMakeFiles/sensor_calibration.dir/Magnetometer.cpp.o.d -o src/lib/sensor_calibration/CMakeFiles/sensor_calibration.dir/Magnetometer.cpp.o -c /Users/kevin/PX4-Autopilot/src/lib/sensor_calibration/Magnetometer.cpp
In file included from /Users/kevin/PX4-Autopilot/src/lib/sensor_calibration/Magnetometer.cpp:36:
/Users/kevin/PX4-Autopilot/src/lib/sensor_calibration/Utilities.hpp:91:2: fatal error: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
        sprintf(str, "CAL_%s%u_%s", sensor_type, instance, cal_type);
        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
1 error generated.
[306/868] Building CXX object src/lib/tecs/CMakeFiles/tecs.dir/TECS.cpp.o
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
message.txt
31 KB

```

Jaeyoung-Lim, 2023-02-14:
@kevinuav Does removing sprintf solve the problrm?

ecmnet, 2023-02-15:
A workaround is to add `set(CMAKE_OSX_SYSROOT "/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk") ` to https://github.com/PX4/PX4-Autopilot/blob/main/CMakeLists.txt

walchko, 2023-02-16:
> A workaround is to add `set(CMAKE_OSX_SYSROOT "/Library/Developer/CommandLineTools/SDKs/MacOSX12.sdk") ` to https://github.com/PX4/PX4-Autopilot/blob/main/CMakeLists.txt

Didn't work for me, still get error:

```
[  6%] Building CXX object platforms/common/CMakeFiles/px4_platform.dir/px4_log.cpp.o
/Users/kevin/Downloads/PX4-Autopilot/platforms/common/px4_log.cpp:106:11: fatal error: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
                        pos += sprintf(buf + pos, "%s", __px4_log_level_color[level]);
                               ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
1 error generated.
make[2]: *** [platforms/common/CMakeFiles/px4_platform.dir/px4_log.cpp.o] Error 1
make[1]: *** [platforms/common/CMakeFiles/px4_platform.dir/all] Error 2
make: *** [all] Error 2
```

mkarklins, 2023-04-16:
@walchko did you ever solve this?

walchko, 2023-04-16:
> @walchko did you ever solve this?

No ... sorry

ryanjAA, 2024-04-12:
Here's how to fix this (and it works on other macOS versions too).

cd into /Library/Developer/CommandLineTools/SDKs/

rename the latest SDK (so if its 14.2 call is 14.1)

then add a symlink to 12.3 but ensure the link is whatever your system is currently using (so if 14.2 do below)

sudo ln -s /Library/Developer/CommandLineTools/SDKs/MacOSX12.3.sdk MacOSX14.2.sdk

12.3 seems to be what works for me.