PX4/PX4-Autopilot #21142: Send Pre-arm setting to ESCs over CAN BUS

TorbjornHouge, 2023-02-17:
## Describe problem solved by the proposed feature
Currently, using PWM ESCs, the autopilot will output a disarmed PWM value (typically 1000 us) once pre-arm button is pressed. This is not sent on DRONECAN, making initialization of some ESCs troublesome since they then require an ARM-DISARM-ARM sequence in order to initialize properly.

## Describe your preferred solution
Introducing a disarmed value for DRONECAN motors (similar to `PWM_MAIN_DISx`), and handling this the same way as for PWM during pre-arm. This value could be named `UAVCAN_EC_DISx`.

## Describe possible alternatives
Have tried configuring ESCs, changing settings so that arming will ensure proper value to ESCs etc, but these solutions all have drawbacks leading to bad performance.

## Additional context
This is close to [this issue](https://github.com/PX4/PX4-Autopilot/issues/20081), but not the exact equal. Still, it could be they should be considered the same.

AlexKlimaj, 2023-02-17:
I have it implemented here. Just needs another pass on this PR to be merged.
https://github.com/PX4/PX4-Autopilot/pull/19748

TorbjornHouge, 2023-02-20:
Excellent, thanks. I hope it passes soon, it is much needed.