PX4/PX4-Autopilot #21144: Ubuntu 22.04 Error reported during installation 404  Not Found

sunxueyi888, 2023-02-17:
root@qk-virtual-machine:/usr/local/goprojects/src/uav/PX4-Autopilot# bash ./Tools/setup/ubuntu.sh
Ubuntu 22.04

Installing PX4 general dependencies
命中:1 http://packages.microsoft.com/repos/code stable InRelease
命中:2 https://packages.microsoft.com/repos/vscode stable InRelease
忽略:3 https://ppa.launchpadcontent.net/fcitx-team/nightly/ubuntu jammy InRelease
获取:4 http://mirrors.ubuntu.com/mirrors.txt Mirrorlist [910 B]
获取:7 http://mirrors.cn99.com/ubuntu jammy-backports InRelease [107 kB]
获取:6 http://mirrors.huaweicloud.com/repository/ubuntu jammy-updates InRelease [119 kB]
获取:8 https://mirrors.bfsu.edu.cn/ubuntu jammy-security InRelease [110 kB]
获取:5 https://mirrors.bupt.edu.cn/ubuntu jammy InRelease [270 kB]
命中:9 https://ppa.launchpadcontent.net/ubuntu-toolchain-r/test/ubuntu jammy InRelease
错误:10 https://ppa.launchpadcontent.net/fcitx-team/nightly/ubuntu jammy Release
  404  Not Found [IP: 185.125.190.52 443]
正在读取软件包列表...
E: 仓库 “https://ppa.launchpadcontent.net/fcitx-team/nightly/ubuntu jammy Release” 没有 Release 文件。
root@qk-virtual-machine:/usr/local/goprojects/src/uav/PX4-Autopilot# 

Please help，3Q

sunxueyi888, 2023-06-08:
> 请问你就解决了吗同学，我也是这个问题

不要用2204的版本

julianoes, 2023-10-20:
Still an issue?

julianoes, 2023-10-20:
I'm going to argue this is not a PX4 issue but an ubuntu download server problem

closed 2023-10-20 by julianoes