PX4/PX4-Autopilot #21146: Servo gimbal range scaling incorrect

dynamicUAVsolutions, 2023-02-17:
Using a 2 axis servo gimbal on a quadcopter on main branch, Ark Fmu-V6x FC with control allocation, dshot output on outputs 1-4 (motors), and PWM 400hz on outputs 5&6 for gimbal servos. 

mnt_mode_in = AUX

Servo throws are way too large compared to vehicle attitude changes while mnt_do_stab is enabled. Limiting output ranges on 5&6 only constrain the over compensation of the servos to a smaller throw distance. Increasing mnt_range_(axis) parameters have some effect in reducing this behavior, but still the gimbal is in an unusable state.

Note: this is not limited to just PWM servos. UAVCAN Servos have the same issue. In the past I have been able to mitigate with the mnt_range adjustments to achieve a workable scale (but not always). 

Due to some servos having different hard internal ranges,
It seems like there should be an actuator scaling parameter for servos.

Issue/Feature request?

Is anyone else experiencing these issues?

AlexKlimaj, 2023-02-17:
Hm, I don't have any experience with the gimbal logic. However, it shouldn't be too difficult to add a scale parameter to the control allocator for gimbals.

RybalkaOleksii, 2024-08-16:
Hi!

@dynamicUAVsolutions did you get any way to fix this issue? I'm facing the same problem, and the setting via mnt_range doesn't suit me, as it still doesn't work to hold the horizon over the full range of the gimbal.