PX4/PX4-Autopilot #21157: Gazebo and jmavsim failing to run MacOS 13.2.1

aerocooper, 2023-02-17:
## Describe the bug
Both Gazebo and Jmavsim are failing to run. I've included a screenshot of the output when I enter the command 
arch -x86_64 make px4_sitl gazebo

I'm on MacOS Ventura 13.2.1 with an M1 chip (which is the reason for including the arch -x86_64 prefix).

I've been beating my head against the wall for a couple days and trying a lot of different things. I've run through a number of different setup processes, so I wouldn't be surprised if something is messy with my config at this point. I just can't figure out what it is. The error is similar if I run the same command for jmavsim.

## Log Files and Screenshots

(base) briancooper@Brians-MacBook-Pro-2 PX4-Autopilot % arch -x86_64 make px4_sitl gazebo 
CMake Error at CMakeLists.txt:129 (list):
  list index: 1 out of range (-1, 0)

CMake Error at CMakeLists.txt:131 (list):
  list index: 2 out of range (-1, 0)

CMake Error at CMakeLists.txt:132 (string):
  string sub-command REPLACE requires at least four arguments.

-- PX4 version: b00efcd966 (b00efcd966..NOTFOUND)
-- Found PythonInterp: /Users/briancooper/opt/anaconda3/bin/python3 (found suitable version "3.9.7", minimum required is "3") 
-- PX4 config file: /Users/briancooper/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- ROOTFSDIR .
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is AppleClang 14.0.0.14000029
-- The C compiler identification is AppleClang 14.0.0.14000029
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Could NOT find gz-transport (missing: gz-transport_DIR)
CMake Warning (dev) at /usr/local/Cellar/cmake/3.25.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.25.2/share/cmake/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:62 (include)
  src/modules/simulation/simulator_mavlink/sitl_targets_gazebo-classic.cmake:2 (find_package)
  src/modules/simulation/simulator_mavlink/CMakeLists.txt:59 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 3.25
-- Found assimp: /usr/local/Cellar/assimp/5.2.5/include (found version "5.2.4") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found DART: /usr/local/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/local/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found Protobuf: /usr/local/lib/libprotobuf.dylib (found version "3.21.12") 
-- Looking for ignition-math6 -- found version 6.13.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreMain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreMain.dylib
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgrePaging.dylib
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreTerrain.dylib
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreProperty.dylib
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreRTShaderSystem.dylib
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreVolume.dylib
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib;debug;/usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/lib/libOgreOverlay.dylib
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-transport8 -- found version 8.4.0
-- Searching for dependencies of ignition-transport8
-- Found ZLIB: /Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/lib/libz.tbd (found version "1.2.11") 
-- Found UUID: TRUE  
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.13.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Looking for ignition-common3 -- found version 3.15.1
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Found DL: TRUE  
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.15.1
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-fuel_tools4 -- found version 4.8.1
-- Searching for dependencies of ignition-fuel_tools4
-- Found CURL: /Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/usr/lib/libcurl.tbd (found version "7.85.0")  
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.9.4
-- Found JSONCPP: TRUE  
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.5
-- Found YAML: TRUE  
-- Checking for module 'libzip'
--   Found libzip, version 1.9.2
-- Found ZIP: TRUE  
-- Looking for ignition-common3 -- found version 3.15.1
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Could NOT find TBB (missing: TBB_DIR)
-- Found gazebo-classic 11.12.0, including sitl_gazebo-classic simulator and gazebo-classic targets
-- Found Java: /Library/Java/JavaVirtualMachines/temurin-19.jdk/Contents/Home/bin/java (found version "19.0.2") 
-- ROMFS: ROMFS/px4fmu_common
-- Configuring incomplete, errors occurred!
See also "/Users/briancooper/PX4-Autopilot/build/px4_sitl_default/CMakeFiles/CMakeOutput.log".
See also "/Users/briancooper/PX4-Autopilot/build/px4_sitl_default/CMakeFiles/CMakeError.log".
Error: /Users/briancooper/PX4-Autopilot/build/px4_sitl_default is not a directory
make: *** [px4_sitl] Error 1

cryptik, 2023-02-21:
Having the same problem trying to build the `make px4_sitl jmavsim` target on Nvidia Jetson Orin, which is also an aarch64 platform.  The error starts here... but not sure what to do about it.

```
CMake Error at CMakeLists.txt:129 (list):
list index: 1 out of range (-1, 0)

CMake Error at CMakeLists.txt:131 (list):
list index: 2 out of range (-1, 0)

CMake Error at CMakeLists.txt:132 (string):
string sub-command REPLACE requires at least four arguments.
```

njanne19, 2023-02-21:
Trying to replicate your bug but haven't gotten it quite yet. For context, I've gotten `px4_sitl` and `jmavsim` to run both by compiling in a regular MacOS terminal using the x86 directive like you've done above (with the `arch` command). I've also gotten success making a default x86 emulated terminal w/Rosetta (https://apple.stackexchange.com/questions/428768/on-apple-m1-with-rosetta-how-to-open-entire-terminal-iterm-in-x86-64-architec#:~:text=Go%20to%20Finder%20%3E%20Applications%20and,verify%20it%20says%20x86_64%20now.)

I also ran into another update in CMakeLists.txt that I needed to make before it would properly build, see these instructions to rule this out as cause: https://github.com/PX4/PX4-Autopilot/issues/20550

I'll circle back if I can replicate

njanne19, 2023-02-24:
Hey Nice! 

I'm really glad you figured out your issue. I don't know how much time you've had with ARM architectures, but have you noticed significant performance hits? I've been running an M1 MacOS setup but exclusively using Rosetta to transpile into x86, would love to know whether or not I can drop this and work with the ARM architecture exclusively. Hoping your linux performance will extend to MacOS

cryptik, 2023-02-24:
So the linux/ARM version is running nicely.  I am going to try and do the similar with a native M1 version and will let you know.  Rosetta is great, but native ARM should be so much faster.  I will let you know how it goes.

duartecdias, 2023-09-03:
Hi, I stumped on this problem while following the PX4 contribution guidelines in https://docs.px4.io/main/en/contribute/git_examples.html

To fix it I followed @cryptik suggestion by fetching the tags from upstream. The way I did it was (1) on my local repository, I make sure I have "upstream" remote set according to the PX4 contribution guidelines; (2) fetch the upstream tags by executing "git fetch upstream --tags".

Just wondering if there is a better way of doing this. Or if there is already updated documentation about this issue and somehow I missed it.

ccw1003, 2023-12-22:
You can try this command:
`git tag -a v1.14.0 -m "PX4 version :1.14.0"`
1.14.0 is your PX4 version.

harunkurtdev, 2024-07-31:
@ccw1003  thank you, it is worked me.