PX4/PX4-Autopilot #21163: SITL with ignition gives build error unknown target

Macavitycode, 2023-02-19:
I was trying to set up PX4 SITL with ignition gazebo, kept running into the unknown target error. I followed [this](http://docs.px4.io/main/en/sim_gazebo_gz/) guide from the official website but kept running into the same set of errors. I also tried running the same command with the following arguments based on what I came across online with the same unknown target result:
- gz_plane
- ignition
- ign_x500
- gz_x500_depth

I had manually installed the files mentioned in `ubuntu.sh` ((this)[https://github.com/PX4/PX4-Autopilot/blob/main/Tools/setup/ubuntu.sh]). I retried the same commands after running `ubuntu.sh`, same error.

```bash
# VERBOSE=1 make px4_sitl gz_x500
Re-run cmake no build system arguments
-- PX4 version: v1.14.0-beta1-389-gfafcdbf4ed (1.14.0)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.6", minimum required is "3") 
-- PX4 config file: /root/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- ROOTFSDIR .
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- The CXX compiler identification is GNU 11.3.0
-- The C compiler identification is GNU 11.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- cmake build type: RelWithDebInfo
-- Could NOT find gz-transport (missing: gz-transport_DIR)
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:72 (include)
  src/modules/simulation/simulator_mavlink/sitl_targets_gazebo-classic.cmake:2 (find_package)
  src/modules/simulation/simulator_mavlink/CMakeLists.txt:59 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 3.05
-- Found fcl: /usr/include (found suitable version "0.7.0", minimum required is "0.5.0") 
-- Found assimp: /usr/include (found version "5.2.0") 
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found version "3.12.4") 
-- Looking for ignition-math6 -- found version 6.13.0
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-transport8 -- found version 8.2.1
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (found suitable version "3.12.4", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.4
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Checking for module 'uuid'
--   Found uuid, version 2.37.2
-- Found UUID: TRUE  
-- Looking for ignition-msgs5 -- found version 5.8.1
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.13.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- Looking for ignition-msgs5 -- found version 5.8.1
-- Looking for ignition-common3 -- found version 3.14.0
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Found DL: TRUE  
-- Found tinyobjloader: /usr/lib/x86_64-linux-gnu/cmake/tinyobjloader/tinyobjloader-config.cmake (found version "2.0.0") 
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.14.0
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-fuel_tools4 -- found version 4.4.0
-- Searching for dependencies of ignition-fuel_tools4
-- Found CURL: /usr/lib/x86_64-linux-gnu/libcurl.so (found version "7.81.0")  
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.9.5
-- Found JSONCPP: TRUE  
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.2.2
-- Found YAML: TRUE  
-- Checking for module 'libzip'
--   Found libzip, version 1.7.3
-- Found ZIP: TRUE  
-- Looking for ignition-common3 -- found version 3.14.0
-- Looking for ignition-math6 -- found version 6.13.0
-- Looking for ignition-msgs5 -- found version 5.8.1
-- Found gazebo-classic 11.10.2, including sitl_gazebo-classic simulator and gazebo-classic targets
-- Could NOT find Java (missing: Java_JAVA_EXECUTABLE Java_JAR_EXECUTABLE Java_JAVAC_EXECUTABLE Java_JAVAH_EXECUTABLE Java_JAVADOC_EXECUTABLE) 
-- ROMFS: ROMFS/px4fmu_common
Architecture:  amd64
==> CPACK_INSTALL_PREFIX = @DEB_INSTALL_PREFIX@
-- Configuring done
-- Generating done
-- Build files have been written to: /root/PX4-Autopilot/build/px4_sitl_default
ninja: error: unknown target 'gz_x500'
make: *** [Makefile:232: px4_sitl] Error 1
```

I'm not sure if I'm missing an environmental configuration step somewhere, but I cannot seem to get SITL working. Please suggest what is to be done, thank you for the help

beniaminopozzan, 2023-02-20:
Hi @Macavitycode , PX4 now supports only `Gazebo Garden`. Unfortunately, the setup script is still installing Ignition Fortress.
The [official guide](http://docs.px4.io/main/en/sim_gazebo_gz/#installation-ubuntu-linux) has the most recent instructions. Follow them, perform a clean build and it will work.

You did not specify which OS you are using and if you have Gazebo Classic installed. Please keep in mind that Gazebo Garden and Gazebo Classic (Gazebo 11) are not compatible. You can have only one of them installed from binaries.

Macavitycode, 2023-02-20:
Hey @beniaminopozzan , thank you for that. The versions I'm using are 
```
Ubuntu 22.04
Ignition fortress
Gazebo 11 (installed when ubuntu.sh was run)
```

I will update to ignition Garden and try again. May I submit a PR to update the install script?

junwoo091400, 2023-02-20:
> May I submit a PR to update the install script or the docs?

Yes please!

Macavitycode, 2023-02-20:
It works once I install `Gazebo Garden`. Thank you @beniaminopozzan .
I will make a PR and submit it soon.

closed 2023-02-20 by Macavitycode

hmer101, 2023-03-07:
Hi guys, I get the same "ninja: error: unknown target 'gz_x500'" error when trying to run 'make px4_sitl gz_x500'. I performed a complete uninstall of ROS Humble and Gazebo Fortress (as well as classic Gazebo) and re-installed ROS2 and Gazebo Garden (using the updated ubuntu.sh) to get to the versions @Macavitycode had above. ROS2 and Gazebo Garden appear to be functioning correctly on their own. I am running on Ubuntu 22.04 and the latest PX4-Autopilot code from the main branch. 

What information do you need from me to help debug this error or am I missing something? Thanks in advance!

idorobotics, 2023-09-21:
I can confirm that this error is still happening. Are there any permanent solutions?

Jaeyoung-Lim, 2023-09-22:
@hmer101 Please post a new issue with a detailed log of your error

PR #21173 (2023-03-01): setup/ubuntu.sh: Only install Gazebo Garden for Ubuntu 22.04
### Solved Problem
- Ingition Fortress was installed instead of Gazebo Garden.
- Deprecated `apt-key` was used in Ubuntu 22.04

Fixes #21163

### Solution
- On Ubuntu 22.04, install only Gazebo Garden
- Key is stored in `/usr/share/keyrings/`