PX4/PX4-Autopilot #21167: Cannot arm the drone with offboard_control using rtps bridge

fbenti, 2023-02-20:
## Describe the bug
I have problem running successfully the offboard_control script on real hardware

## Drone :
- Holybro v5

## To Reproduce
Steps to reproduce the behavior:
On pixhawk:
1. I flashed the pixhawk from the stable `release/1.13` setting the `CONFIG_MODULES_MICRODDS_CLIENT=n
` parameter in `~/PX4-Autopilot/boards/px4/fmu-v5/rtps.px4board`
2. I set the `RTPS_CONFIG` parameter to `TELEM2`, and then `SER_TEL2_BAUD` to `921600 8N1`.

On pc:
1. I build the px4_ros_com workspace from the `release/1.13` branch with the following output.
[output(2).txt](https://github.com/PX4/PX4-Autopilot/files/10782250/output.2.txt)
To notice that the building always gives at the end ` 1 package had stderr output: px4_ros_com` but without saying where  the error comes from.
2. Connect `TELEM2` and my pc through serial.
4. run rtps agent with `micrortps_agent -t UART -d /dev/ttyUSB0 -b 921600` getting the following output
```
--- MicroRTPS Agent ---
[   micrortps_agent   ]	Starting link...
[   micrortps_agent   ]	UART transport: device: /dev/ttyUSB0; baudrate: 921600; poll: 1ms; flow_control: No
---   Subscribers   ---
- DebugArray subscriber started
- DebugKeyValue subscriber started
- DebugValue subscriber started
- DebugVect subscriber started
- OffboardControlMode subscriber started
- OpticalFlow subscriber started
- PositionSetpoint subscriber started
- PositionSetpointTriplet subscriber started
- TelemetryStatus subscriber started
- Timesync subscriber started
- VehicleCommand subscriber started
- VehicleLocalPositionSetpoint subscriber started
- VehicleTrajectoryWaypoint subscriber started
- OnboardComputerStatus subscriber started
- TrajectoryBezier subscriber started
- VehicleTrajectoryBezier subscriber started
- TrajectorySetpoint subscriber started
- VehicleMocapOdometry subscriber started
- VehicleVisualOdometry subscriber started
-----------------------

----   Publishers  ----
- Timesync publishers started
- TrajectoryWaypoint publisher started
- VehicleControlMode publisher started
- VehicleOdometry publisher started
- VehicleStatus publisher started
- CollisionConstraints publisher started
- TimesyncStatus publisher started
- SensorCombined publisher started
- VehicleTrajectoryWaypointDesired publisher started
-----------------------
```
5. Now, if I run the `sensor_combined_listener`, I correctly get the data, however if I run the `offboard_control`, I get the following output but I cannot arm the drone.
```
Starting offboard control node...
[INFO] [1676882384.131341654] [offboard_control]: Arm command send
```

## Additional context
Add any other context about the problem here.

beniaminopozzan, 2023-02-20:
Hi, Could you share the flight log?

fbenti, 2023-02-21:
Silly mistake. By default, the remote controller has to be connected to the px4, otherwise offboard control is not allowed.  
Anyway I have a couple of questions.
- Is the flight log  the ulg file that is saved of the sd card? If so, the only way to analyse it is the [Flight Review](http://logs.px4.io/ ) or are there other ways?
- I know it is not recommended, but if I want to deactivate that by default the remote controller has to be connected, which parameter should I change?
- If you had a look at the build output of the  px4_ros_com workspace, do you have an idea from where the `1 package had stderr output: px4_ros_com` comes from?

closed 2023-02-21 by fbenti

beniaminopozzan, 2023-02-23:
Hi @fbenti ,
you are looking for [COM_RCL_EXCEPT](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#COM_RCL_EXCEPT)  :):)

You may also be interested in [NAV_DLL_ACT](https://docs.px4.io/main/en/advanced_config/parameter_reference.html#NAV_DLL_ACT)