PX4/PX4-Autopilot #21179: microdds client: add Agent IP as PX4 parameter

beniaminopozzan, 2023-02-22:
## Describe problem solved by the proposed feature
Differently from _mavlink_, when UDP transport protocol is used, the _microdds client_ requires the explicit IP address of the companion computer running the  _agent_.
This value is currently passed to the module as an argument at startup in the form
```sh
microdds_client start -t udp -p 8888 -h 192.168.0.1
```
which tells that UPD communication to an Agent running at `192.168.0.1` and listening to the port `8888` should be used.
If not given, the default IP value is the loopback address `127.0.0.1` which is just right for simulations.
However, when working on real drones this prevents starting the client from PX4 parameters ( `XRCE_DDS_0_CFG=ethernet`) as there is no PX4 parameter for storing the Agent IP address. As a consequence, the default value is always used.

https://github.com/PX4/PX4-Autopilot/blob/2008a447c352fb04d9b817ec01593c66a92d7bae/src/modules/microdds_client/module.yaml#L14-L33

## Describe your preferred solution
It would be great to have the Agent IP address as a PX4 parameter. One INT32 parameter encoding the address would be enough and the client would just have to parse it.

## Describe possible alternatives

- Instead of using 1 parameter, a more user friendly solution could be to use 4 separate parameters to encode the address as PARAM1:PARAM2:PARAM3:PARAM4.
- Implement _string_ parameters?
- Remove the `supports_networking: true` from `XRCE_DDS_CFG` and force the users to use [system startup](https://docs.px4.io/main/en/concept/system_startup.html) if they want to use UDP transport protocol.

hamishwillee, 2023-02-22:
@dagar @bkueng @mrpollo This would allow using PX4 parameters to configure a ROS2 connection over Ethernet. 
A bit of a glaring omission.

mrpollo, 2023-02-22:
@davids5 can we use this macro to convert from Int32 to IP? https://github.com/PX4/NuttX/blob/master/include/nuttx/net/ip.h#L287-L303

bkueng, 2023-02-23:
We can use a single int32 parameter and extend the metadata so the GCS can display it as IP.
Or simpler yet, as most setups follow the same network setup, querying the network gateway could work to get the IP automatically.

davids5, 2023-02-23:
@mrpollo we could use it. But the idea of using params for IP addresses has been rejected in the past. @dagar argued against it for UI, use and consistency issue, 

The simple format `#  define DEFAULT_IP      0XC0A80003  // 192.168.0.3`  is understandable but not as simple as decimal dot notation. QGC and param would need to support it.

At the time `netman` was added to store the network interface settings. it uses "ipcfg_{read|write}" and stores the data in eeprom. The settings can be changed by placing a `net.cfg` file on the SD card.  [see](https://github.com/PX4/PX4-Autopilot/blob/55563eba49f87624d7b73356ea3d421fee7ec081/src/systemcmds/netman/netman.cpp#L404-L427)

beniaminopozzan, 2023-02-24:
Hi @bkueng , with

> querying the network gateway could work to get the IP automatically.

you mean that the network gateway and the companion computer are the same machine (same IP)?

bkueng, 2023-02-24:
No, if the companion provides DHCP to the FMU, it should also set its own IP as gateway.

ozitoune, 2023-02-27:
I have tried without gateway, and for now i can't make it work :( i not understand why we can't make peer to peer connexion without gateway... nobody will put a router in his drone !!  ,So the connexion will be always peer to peer between a pixhawk  and a raspberry (or jetson or other companion) 
So parameters on px4 should be only the ip  adress off the pixhawk and the ip  adress off the companion!!  And thre client have to start with the -h "ip off the companion" automaticly.

beniaminopozzan, 2023-02-28:
@bkueng I see, but this solution would work only in that specific solution, which is not the default one suggested in the documentation https://docs.px4.io/main/en/advanced_config/ethernet_setup.html#ubuntu-ethernet-network-setup

Would modifying param and QCG for allowing IP-type parameters be that problematic? If so, what about extending `netman` with `COMPANION_COMPUTER_IP`-like parameter? Then microdds could fetch it. Is it a reasonable compromise?

bkueng, 2023-02-28:
> @bkueng I see, but this solution would work only in that specific solution, which is not the default one suggested in the documentation https://docs.px4.io/main/en/advanced_config/ethernet_setup.html#ubuntu-ethernet-network-setup

It should work if we change to `ROUTER=192.168.0.1`. Using the broadcast IP could also work.

> Would modifying param and QCG for allowing IP-type parameters be that problematic? If so, what about extending netman with COMPANION_COMPUTER_IP-like parameter? Then microdds could fetch it. Is it a reasonable compromise?

Also fine with me. Do you want to do the QGC UI changes?

ozitoune, 2023-02-28:
I not understand why is problematic to have a parameter for the companion computer address ? 

>  @dagar argued against it for UI, use and consistency issue,

I don't see any problem to have the possibility to define the companion IP address by UI, for me it's the better and simplest solution. If someone can explain why is not a good choice or not possible ?

dagar, 2023-02-28:
> I not understand why is problematic to have a parameter for the companion computer address ?
>
>
> I don't see any problem to have the possibility to define the companion IP address by UI, for me it's [...]

We don't need to fully get into it here, but just for background context it was more that I was opposed to forcing the system's network configuration into the parameter system when for actual products we primarily want a static IP address (corresponding to your vehicle's embedded flight controller + linux companion) with the occasional option to temporarily switch over to DHCP for convenience on your home network. I specifically didn't want to unnecessarily open that hole where people were reconfiguring their network connection (often over that network connection) and potentially throwing away a static config necessary for the system to function when all they actually wanted/needed to do was switch to DHCP temporarily.

Secondly if it can be avoided I don't particularly like cramming things like an IP address into an INT32 parameter because internally the system doesn't have the metadata so it can be awkward to work with (eg param show, param set, etc). Adding proper string parameters within PX4 actually isn't that hard, but Mavlink legacy gets in the way. Anyway, for now it seems harmless enough in this case.

swimmingseeds, 2023-03-01:
This issue has been mentioned on **Discussion Forum for PX4, Pixhawk, QGroundControl, MAVSDK, MAVLink**. There might be relevant details there:

https://discuss.px4.io/t/px4-community-q-a-march-01-2023/30941/6

swimmingseeds, 2023-03-01:
This issue has been mentioned on **Discussion Forum for PX4, Pixhawk, QGroundControl, MAVSDK, MAVLink**. There might be relevant details there:

https://discuss.px4.io/t/px4-community-q-a-march-01-2023/30941/1

mrpollo, 2023-03-01:
FYI we are moving ahead with the INT32 solution for now, and will look into adding strings as parameters later to actually complete the implementation to QGC

beniaminopozzan, 2023-03-03:
Could someone with a compatible board give a try to the PR?
It works in simulation but without a board I cannot test it further

Ryanf55, 2023-03-22:
> ## Describe possible alternatives
> * Instead of using 1 parameter, a more user friendly solution could be to use 4 separate parameters to encode the address as PARAM1:PARAM2:PARAM3:PARAM4.
> * [...]

You could also try using discovery
https://micro-xrce-dds.docs.eprosima.com/en/latest/client_api.html#discovery-api

PR #21234 (2023-03-27): microdds_client: support Agent IP address as PX4 parameter
### Features Implemented

1. Removed incomplete multi instance capability for microdds_client.
2. Added new parameter `XRCE_DDS_AG_IP` that encodes the micro XRCE-DDS Agent IP. Its default value corresponds to `127.0.0.1`.
3. When the client is started without specifying the udp port (missing `-p` argument) then the parameter `XRCE_DDS_PRT` is used.
4. When the client is started without specifying the Agent IP (missing `-h` argument) then the parameter `XRCE_DDS_AG_IP` is used.
5. Increased verbosity of `microdds_client status`.
6. To  [... cut]
closed 2023-03-27 by mrpollo