PX4/PX4-Autopilot #21184: COM_RC_IN_MODE=1 won't disable RC input pre-arm checks

riosnav, 2023-02-23:
## Describe the bug
Using a Pixhawk 4 with an older firmware version (1.12.3) would allow to arm the vehicle without any RC input setting the parameter COM_RC_IN_MODE to 1 (Joystick/No RC Checks).

I am now setting up a Pixhawk 6C for the same application, running FW version 1.13.2. After setting the parameter COM_RC_IN_MODE to 1 (Joystick Only), QGround control stops complaining about Radio not being configured (the icon turns from red to gray as expected), but when trying to arm the device I get a "Critical: Arming Denied! manual control lost"

## To Reproduce
Steps to reproduce the behavior:
1. Set COM_RC_IN_MODE to 1 without any RC input
2. Try to arm the vehicle

## Expected behavior
The RC signal lost check should have been fully disabled after setting parameter COM_RC_IN_MODE to 1.

## Log Files and Screenshots
There is none

## Drone (please complete the following information):
- Holybro Pixhawk 6C.

## Additional context
The error seems to be triggered by line 76 at [preArmCheck.cpp](https://github.com/PX4/PX4-Autopilot/blob/8deebd07b83907518b4ccb127f49affa079c98e9/src/modules/commander/Arming/PreFlightCheck/checks/preArmCheck.cpp#L76). I'm honestly not sure wether this is an actual bug or ti's the new expected behaviour for safety reasons, but it was definitely useful to have the ability to arm the device without RC input. We have an application in which we are using the Pixhawk basically as a logger with no actuation capabilities, and we use the arm/disarm commands as a way to start and stop SD logging.

sfuhrer, 2023-02-28:
@MaEtUgR can you pitch in?

MaEtUgR, 2023-05-23:
This was fixed for 1.14 with https://github.com/PX4/PX4-Autopilot/pull/20695. I wasn't aware of an issue on 1.13 👀