PX4/PX4-Autopilot #21228: CUAV X7Pro Flash Overflow on main

mrpollo, 2023-03-01:
We gotta fix this in main branch, and make sure it doesn't slip into the v1.14 release branch

https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621

```
NuttX/nuttx/libs/libc/libc.a  NuttX/nuttx/sched/libsched.a  -lm  -lgcc && :
[1216](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1217)
Memory region         Used Size  Region Size  %age Used
[1217](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1218)
        ITCM_RAM:          0 GB        64 KB      0.00%
[1218](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1219)
           FLASH:     1966132 B      1920 KB    100.00%
[1219](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1220)
       DTCM1_RAM:          0 GB        64 KB      0.00%
[1220](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1221)
       DTCM2_RAM:          0 GB        64 KB      0.00%
[1221](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1222)
        AXI_SRAM:       44360 B       512 KB      8.46%
[1222](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1223)
           SRAM1:          0 GB       128 KB      0.00%
[1223](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1224)
           SRAM2:          0 GB       128 KB      0.00%
[1224](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1225)
           SRAM3:          0 GB        32 KB      0.00%
[1225](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1226)
           SRAM4:          2 KB        64 KB      3.12%
[1226](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1227)
          BKPRAM:          0 GB         4 KB      0.00/opt/gcc/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: cuav_x7pro_default.elf section `.data' will not fit in region `FLASH'
[1227](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1228)
/opt/gcc/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 52 bytes
[1228](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1229)
collect2: error: ld returned 1 exit status
[1229](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1230)
%
[1230](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1231)
ninja: build stopped: subcommand failed.
[1231](https://github.com/PX4/PX4-Autopilot/actions/runs/4304244468/jobs/7505521621#step:5:1232)
Error: Process completed with exit code 2.
```

swimmingseeds, 2023-03-01:
This issue has been mentioned on **Discussion Forum for PX4, Pixhawk, QGroundControl, MAVSDK, MAVLink**. There might be relevant details there:

https://discuss.px4.io/t/px4-community-q-a-march-01-2023/30941/1

junwoo091400, 2023-03-25:
Now in the `main` it doesn't overflow anymore: https://github.com/PX4/PX4-Autopilot/actions/runs/4516810959/jobs/7955512240

closed 2023-03-25 by junwoo091400