PX4/PX4-Autopilot #21241: Vehicle_attitude_setpoint cannot be reset when switching from attitude control back to velocity control

calrho, 2023-03-03:
I am using the px4 (1.14beta) -> ROS2 bridge to control a quad. I can control it using the trajectory_setpoint ROS2 message and I can also control it using the vehicle_attitude_setpoint message. However, there seems to be an issue when switching between the two. It seems as if setting the vehicle_attitude_setpoint irrevocably sets the target attitude for the quad and even if you no longer send this message and start sending velocity commands (setting the appropriate fields in the offboard_control_mode topic), that the quad is stuck with the previously sent atti command and PX4 will not start generating its own based on the new velocity commands. Is this a bug or an intentional feature in the offboard control logic? Or am I setting something else wrong?

Jaeyoung-Lim, 2023-03-03:
@calrho Do you have a log?

calrho, 2023-03-03:
@Jaeyoung-Lim In getting the log I realised my mistake, although the outcome is still a little bizarre. So I had not set the timestamp in the trajectory_setpoint message (i.e. 0) but I had set it in the attitude_setpoint message. 

What I find strange is that offboard control initially worked without the timestamp on vel, but then after sending atti with the updated timestamp, resending the same 0 timestamp vel message after would not kick it back into velocity control. Either way my issue is solved and I thank you for managing to poke me in the right direction. Potential lesson to others to check timestamps!

closed 2023-03-06 by calrho