PX4/PX4-Autopilot #21248: Gazebo exits with exit code 256 when bypassing MC_Rate_Control

karishmapatnaik, 2023-03-06:
I am trying to directly publish to the actuator_controls_0 from the attitude controller bypassing the rate controller. After the code changes, the build seems fine. However, when I start the gazebo simulator, it exits with code 256. I edited the following configuration file `default.px4board` under `/boards/px4/sitl` to remove  `CONFIG_MODULES_MC_AUTOTUNE_ATTITUDE_CONTROL=y` and `CONFIG_MODULES_MC_RATE_CONTROL=y`

I also changed `rc.mc_apps` to remove both the modules for rate control and the autotune attitude control. 

Any help to where I can take a look is much appreciated, thanks! Here is a screenshot of the error:

[image]

Jaeyoung-Lim, 2023-03-06:
@karishmapatnaik Have you tried disabling lockstep?

If your actuator setpoints are not published in sync with the sensor messages, the simulation will lose sync with PX4 and stall.

karishmapatnaik, 2023-03-06:
@Jaeyoung-Lim Thank you for helping, yeah I had disabled the lockstep, it doesn't work. I will look into the synchronization, but is there any other configuration setting I am missing apart from the ones I set? I hope the settings are correct to bypass this module?   l shall test the message timestamps meanwhile, thank you.